c3d/db48x #1401: Equation Library: close analysis of bug c#2 in "Bernoulli Equation"

Wiljea, 2024-12-28:
Here we encounter the "Inconsistent units" error, see below:
```rpl
P2=25_psi  P1=75_psi  y2=35_ft  y1=0_ft  D1=18_in  ρ=64_lb/ft^3  v1=100_ft/s
@ Failing [ A1=254.46900 4941 in↑2 ΔP=-50. psi Δy=35. ft Q=10 602.87520 59 ft↑3/min M=678 584.01317 5 lb/min v2=122.42131 1569 ft/s A2=207.86332 19 in↑2 D2=16.26836 81217 in ]
@ C#3 NOT OK. MSOLVER: "Inconsistent units", SOLVE: OK for A1;ΔP;Δy;Q;M calculated one at a time in order; NOT OK for v2;A2;D2:
@ "Inconsistent units" while searching for each unknown. It seems that isolating v2 from implicit eqn (1) doesn't work here
'ROOT(ⒺBernoulli Equation;[A1;ΔP;Δy;Q;M;v2;A2;D2];[1_in^2;1_psi;1_ft;1_ft^3/min;1_lb/min;1_ft/s;1_in^2;1_in])'
```

Wiljea, 2024-12-28:
Alternate presentation adding one more known value: `v2`
```rpl
P2=25_psi  P1=75_psi  y2=35_ft  y1=0_ft  D1=18_in  ρ=64_lb/ft^3  v1=100_ft/s  v2=122.421311569_ft/s
@ Failing [ A1=254.46900 4941 in↑2 ΔP=-50. psi Δy=35. ft Q=10 602.87520 59 ft↑3/min M=678 584.01317 5 lb/min v2=122.42131 1569 ft/s A2=207.86332 19 in↑2 D2=16.26836 81217 in ]
@ C#3 NOT OK. MSOLVER: "Inconsistent units", OK for A1;ΔP;Δy;Q;M solved one at a time NOT OK for A2;D2 wrong values @ are obtained which is strange because eqn (7) should resolve for A2 BUT the value of Q change inadvertently, why?
'ROOT(ⒺBernoulli Equation;[A1;ΔP;Δy;Q;M;v2;A2;D2];[1_in^2;1_psi;1_ft;1_ft^3/min;1_lb/min;1_ft/s;1_in^2;1_in])'
```

c3d, 2025-01-19:
The root cause for this one seems to be an actual `Inconsistent units` caused by
the `=0` at the end of the definition of the Bernouilli equation:

```
'(ΔP_Pa)/(ρ_(kg/m^3))+((v2_(m/s))^2-(v1_(m/s))^2)/2+Ⓒg*(Δy_m)=0'
```

This fails when the left-hand-side of the equation cannot be reduced to a
unit-less value, e.g.:

```
-2 283.86297 50002 62206 79154_m·ft/s↑2
```

Wiljea, 2025-02-06:
Done.

closed 2025-02-06 by Wiljea