c3d/db48x #1402: Equation Library: close analysis of bug c#4 in "Flow In Full Pipes"

Wiljea, 2024-12-28:
Error condition: "Inconsistent units":
```rpl
ρ=62.4_lb/ft^3  D=12_in  vavg=8_ft/s  P2=15_psi  P1=20_psi  y2=40_ft  y1=0_ft  μ=0.00002_lbf*s/ft^2  ΣK=2.25  ϵ=0.02_in  L=250_ft
@ Failing [ ΔP=-5. psi Δy=40. ft A=113.09733 5529 in↑2 n=1.03121 95050 1⁳⁻⁵ ft↑2/s Q=376.99111 8431 ft↑3/min M=23 524.24579 01 lb/min Reynolds=775 780.51628 2 f=4.32637 07915 4⁳⁻³  W=24.95162 12864 hp ]
@ C#4 NOT OK. MSOLVER: "Inconsistent units". SOLVE hallucinate ΔP; "Constant?" for Δy, A, Q, M ; OK for n & f; "Inconsistent units" for W
'ROOT(ⒺFlow In Full Pipes;[ΔP;Δy;A;n;Q;M;Reynolds;f;W];[1_psi;1_ft;1_in^2;1_ft^2/s;1_ft^3/min;1_lb/min;1;1;1_hp])'
```

c3d, 2025-01-19:
The core problem for the Flow functions is fixed by simplification of scalar
unit objects, e.g. `1_km/mi`.

Wiljea, 2025-02-06:
Done.