c3d/db48x #1407: Excessive dead space on iPhone

marsell, 2025-01-16:
Hello,

I am using db48x on an iPhone. I prefer smaller phones, so it's an iPhone SE, which means that the screen is also small.

When I opened up db48x, the calculator was notably smaller than I expected, before I realized that it has a border and other dead space. To wit:

http://desacci.org/db48x-deadspace.png

The dead space is outlined in red.

Perhaps it's not possible to get rid of the actual status line, but the rest is wasted space on a screen that cannot afford wasted space.

Is there anything I can currently do to remedy this? Or can an option be added to remove most/all of what's currently outlined in red?

c3d, 2025-01-17:
It is a bit difficult to adjust the sides while preserving the aspect ratio (which is necessary to accurately identify the keys when touching). 

[image]

The best I could do is move the ⓘ sign inside the screen instead of outside of it, which makes it possible to slightly move things.

* The space at the bottom is a result of the aspect ratio computation
* The limiting factor is the combination of the total height, available space and aspect ratio constraint.

So the best I can get is something that is very slightly wider, at the expense of having the ⓘ button outside of the screen. I'm frankly not sure that I like it (it looks worse on all more modern iPhones).

[image]

marsell, 2025-01-20:
I'll take whatever I can get. If you make the "ⓘ" inside/outside as an option, preserving the current behaviour, then maybe that'd be acceptable to most people?

Out of curiosity, what is the logic for height, width and aspect ratio? It's not immediately obvious why there'd be empty margins on both the side and bottom if it's just to preserve aspect ratio.

c3d, 2025-11-08:
Indeed, preserving aspect ratio and respecting the area that Apple marks as "safe" are what guided the implementation.

For a few releases, I tried to enlarge things to the whole drawable area, but the problem with that is that this leads to side pixels becoming invisible on iPhone 13 and later. They are rendered and are shown while you are in the application switcher, so it's really a matter of Apple intentionally zooming out some unspecified number of pixels on the borders of the application.

We are back within the "safe" area now. I would appreciate another screen capture with the latest release.

marsell, 2025-11-10:
[image]