c3d/db48x #1410: Fault using degree °

Uli-Moin, 2025-01-25:
9
sin
cos
tan
tan^-1
cos^-1

Fault: inconstant units

Not rally a fault. But can I switch off the degree ° symbol?

prismv, 2025-01-26:
In the Conv Menu use Value (F3) to get only the value without the ° unit (assuming that DEG is set).

This is the whole sequence in one line with substraction of 
```
«
	Deg 9 Sin Cos Tan Atan UVal Acos UVal Asin 9_° -
» Eval
```

The result depends on the `Prec` setting. If this is e.g. set to 48, I get 5.3183x10⁻⁴¹ ° as result.

Uli-Moin, 2025-01-26:
Many thanks. That is a solution. And I did not know that.

Is there no way to switch off the degree (°) as an unit to avoid to use the Value klick?

Uli-Moin, 2025-01-26:
> In the Conv Menu use Value (F3) to get only the value without the ° unit (assuming that DEG is set).
> 
> This is the whole sequence in one line with substraction of 9° to get the deviation as test result:
> 
> ```
> «
> 	Deg 9 Sin Cos Tan Atan UVal Acos UVal Asin 9_° -
> » Eval
> ```
> 
> The result depends on the `Prec` setting. If this is e.g. set to 48, I get 5.3183x10⁻⁴¹ ° as deviation to exactly 9°.

Many thanks. That is a solution. And I did not know that.

Is there no way to switch off the degree (°) as an unit to avoid to use the Value klick?

c3d, 2025-01-27:
> Not rally a fault. But can I switch off the degree ° symbol?

Yes, use the `NoAngleUnits` setting. But be careful, in the sense that having the angle unit guarantees you get the correct result if you do something like `9 sin asin rad sin`. With the flag set, you get 0.412118485242, which is the sine of 9 radians and not 9 degrees. With the default setting for the flag, you get 0.15643446504.

As indicated by @prismv `UVAL` can be used to remove any unit, in case what you need is the numerical value.

I will close this issue, please feel free to reopen if necessary.

closed 2025-01-27 by c3d