c3d/db48x #1415: Flashing Error

orome, 2025-02-05:
For 0.9.0 on a DM42 I get 

```
File CRC: FAIL

Flashing error:
File doesn't have required CRC
Press any key
```

orome, 2025-02-05:
I also cant load the program (can't navigate to it on the file browser, where I can only go up and down can't select any files), so I'm stuck in the menus and can't access calculator functionality at all.

c3d, 2025-02-06:
I suggest you download the file again from https://github.com/c3d/db48x/releases/download/v0.9.0/db48x-v0.9.0.tgz, then extract it directly in the calculator. I just extracted it and it worked. It's possible there was simply a glitch writing the files the first time (I had that a couple fo times during the development).

Also check the md5sum for the file:

d41d8cd98f00b204e9800998ecf8427e  db48x-v0.9.0.tgz

orome, 2025-02-06:
That passes CRC and proceeds through reboot but after that it says QSPI not loaded and still shows 0.8.11 as the version.