c3d/db48x #1418: problems with variable set as constant in equation library example "de Broglie wave"

prismv, 2025-02-07:
version: 0.9.0

The "de Broglie wave" equations contain variable `m_kg` that is set to `m=Ⓒme` in doc example. Root is solved without problems, but solving for m gives error "SolvingMenuSolve error: Bad argument type". Furthermore change of m with e.g. `me*2` results in `m='me*2' kg`, which would lead to an solver error.

c3d, 2025-03-09:
Hi @prismv, thanks for the report. I can't reproduce on 0.9.2. Could you please either give me an exact sequence of keys that leads to the problem, or alternatively close the issue if it's indeed fixed?

prismv, 2025-03-09:
No, it's _not_ fixed in 0.9.2 Why should it be fixed? 

How to reproduce:

1. Populate variables by selecting example from documentation -> this sets i.a. `m=Ⓒme` 
2. Enter then Solve menu with this equation system, e.g. right shift  F5 'de Broglie wave' equation menu entry.
3. Solve all missing variables λ, K, v
4. _now_ solve for m 
    ⇒ SolvingMenuSolveError
5. _now_ assign result of `Ⓒme 2 x` to mass m 
   ⇒ solve for any variable related to m will not work anymore because of a unit problem with `m='me*2' kg`

c3d, 2025-06-07:
The example on the HP Museum Forum finally allowed me to understand what you were doing exactly. The problem is now fixed, both for the [original case on the HP Museum Forum](https://www.hpmuseum.org/forum/thread-20157-post-205095.html#pid205095) and for the case described here, which I can now reconstruct from your description here and the protocol that I finally understood.

prismv, 2025-06-07:
> [Verfolgungsbild][https://github.com/notifications/beacon/ACLSOQDXKEMZL2HCD6SCX6L3CNDUJA5CNFSM6AAAAABWVWNYRCWGG33NNVSW45C7OR4XAZNMJFZXG5LFINXW23LFNZ2KUY3PNVWWK3TUL5UWJTVQALZGQ.gif]

Great. It was sadly not straightforward for me to produce the easiest case as given on MoHPC.

prismv, 2025-06-08:
Issue turns out to describe 2 technical 'bugs'.