c3d/db48x #1421: CustomMenu as default for VAR key?

orome, 2025-03-02:
Shouldn't the `CustomMen`u be the first rather than the second thing that the `[VAR]` key brings up? When `CustomMenu` is defined, it's odd that it takes two clicks to bring it up and that on the way one sees variables, cluttered with `CustomMenu`. If a user has gone through the trouble to create a custom menu, I think they'd expect easier access to it, with all the currently defined variables at the next level.

c3d, 2025-03-09:
Good idea. I will add a flag for that.

c3d, 2025-03-09:
I wonder what would be more practical:
1. A flag that applies globally as a user preference
2. A way for `CustomMenu` to indicate it wants to be shown first

See [this comment](https://github.com/c3d/db48x/issues/1417#issuecomment-2709020944) as to why it may matter for the equation library.

orome, 2025-03-09:
> I wonder what would be more practical:
> 
> 1. A flag that applies globally as a user preference
> 2. A way for `CustomMenu` to indicate it wants to be shown first
> 
> See [this comment](https://github.com/c3d/db48x/issues/1417#issuecomment-2709020944) as to why it may matter for the equation library.

I'm not sure I understand #1417, but from a naive UX perspective I think 2 is preferred: having a way for `CustomMenu` to indicate it wants to be shown first.