c3d/db48x #1422: RPL Geometry

whoffmans, 2025-03-06:
RPL Geometry?
-----------------------
Hi all,

The other day I was thinking about the ability to do geometry / geometric calculations on calculators, and if this in fact could be integrated into RPL as well, so that it would be really an integral part of the calculator instead of some isolated app.
I put my thoughts on paper to give you a clearer idea of what I mean. See attached document.

I am very interested to know what you think about this idea. I can't imagine that it is new, because it seems a bit obvious. But I could not find existing info about it.

Thanks,
Werner

[RPL Geometry v01.pdf](https://github.com/user-attachments/files/19111261/RPL.Geometry.v01.pdf)

c3d, 2025-03-09:
Thanks a lot for sharing, @whoffmans. This is indeed a really interesting addition and a well though-out overview of the problem. 

Here are some reactions after a quick reading of the document.

1.1.1. Objects: RPL manipulates data through lists. I don't see a real need for new object types, and I'm afraid it might reduce flexibility.

A convention for structured objects could be to have a list that begins with a name. This could give additional flexibility in specifying objects, like: `{ POINT 0 0 }`, `{ LINE { POINT 0 0 } { VECTOR 1 0 } }` vs. `{ LINE { VECTOR 1 0 } { POINT 0 0 } }` vs . `{ LINE { POINT 0 0 } { POINT 1 0 } }`, and so on. This makes the analysis of the line a bit more complicated, but it means you can build your object using existing tools like `+` on lists.

1.1.2. The commands in question could very well generate the above objects

2.1. Triangle identities: they are given as an example for the multiple equation solver. However, it would be a good idea to add them to the equation library.

2.2.1. Some exact values are already taken into account, but not all in the table. Could be an interesting and relatively easy addition to the existing infrastructure for trigs.

evgaster, 2025-05-13:
If  [c3d](https://github.com/c3d) would consider including a [CAS](https://www-fourier.univ-grenoble-alpes.fr/~parisse/english.html)  then a lot of [geometry](https://xcas.univ-grenoble-alpes.fr/en.html#Fonctionnalit%C3%A9s) stuff comes with it along with lots of other goodies.

Including a CAS is probably quite a challenge. 

Looking at "A convention for structured objects ..." that doesn't match the postfix paradigm. Wouldn't it be better to have {  { 0 0 point } { 1 0 vector } line } then the prefix example? In my mind it would make implementing some of the geometry functionality in rpl easier.