c3d/db48x #1426: Issues with bivariate statistics variables

prismv, 2025-03-09:
I tried to implement as RPL expressions standard deviation of linear regression results as `sLR` vector. 
Furthermore In the HP50g user guide page 18-52/-53 there are formulas of confidence intervals - `cfLR` is the vector name here.

Input was merely possible as a direct expression (or more precise multiplication of 2 expressions), because e.g. `(ΣSize)` is needed in input mode for proper evaluation, but the brackets are removed on saved expression on stack. There seems to be also a clash with  ΣX² and (ΣX)².

The workaround was to use local variables for the dubious acting parameters.
The following shown implementation works for `cfLR` without any bad indications, but `sLR` gets a wrong output every second time `Correlation` is evaluated in alternation!

``` rpl
	dStats
		Directory {
			cfLR
				«
					'ΣSize' 'ΣX' 'ΣX²'
					'ΣY²' 'Correlation'
					→ N SX SXX SYY
					r
					«
						'√((N-1)÷(N-2)·SYY·(1-r²))'
						[ '√((N+SX²÷SXX)÷N²)'
							'1÷√ SXX' ]
						× →Num
					»
				»
			sLR
				«
					'ΣSize' 'ΣX' 'ΣY'
					'ΣXY' 'ΣX²' 'ΣY²'
					→ N SX SY SXY SXX
					SYY
					«
						'√((SYY-Slope·SXY-Intercept·SY)·SXX÷(N-2)÷(N·SXX-SX²))'
						[ 1 '√(N÷SXX)'
						] × →Num
					»
				»
			ΣParameters
				{ 1 2 49/5 1 LinearFit
				}
			ΣData
				[[ 1 11 ]
				  [ 2 19 ]
				  [ 3 33 ]
				  [ 4 39 ]]
		}
```

Because of several other reference implementations I can be sure that the result of `sLR` has to be `[ 2.846 1.0392 ]`, so Slope is `49/5` with std.dev. of `1.0392`. 

(If the result of `cfLR` with `[ 6.838 1.8441 ]` is an expected result is still open for me to be checked).

c3d, 2025-03-13:
Thanks for the resport, @prismv. I will look into it.

prismv, 2025-04-06:
@c3d with 0.9.3 even my workaround is broken. The state file gives Syntax error at position `'ΣY²` of the top function cfLR.

prismv, 2025-04-09:
As written in [forum post](https://www.hpmuseum.org/forum/thread-20157-post-202006.html#pid202006) the mitigation is to use `ΣY²` or `ΣY2`. I see with current dev version that the input of ΣY² via catalog menu into `''` is converted properly to `ΣY2` according to ids.tbl ALIAS definition.

The only mystery (for me) is that the input of `'ΣY²'` by typing of `''`, left cursor, `Σ`, `Y` and `²` gives syntax error, but in case of `'ΣX²'` there is no error.

c3d, 2025-04-11:
OK, I finally get to this particular bug.

Commands like `ΣX²` are not supported algebraic commands on the HP50G.
As an extension, DB48x accepts them, but indeed, it fails to parse `ΣY²` correctly because the square operator is treated specially in parsing.
I will fix that.

That being said, I don't see the point of quoting the values you pass to local variables. Your goal is to have their values anyway, right? So the following is a more efficient way to write the same code:

```rpl
Directory {
	cfLR
		«
			ΣSize ΣX ΣX² ΣY² Correlation
			→ N SX SXX SYY r
			«
				'√((N-1)÷(N-2)·SYY·(1-r²))'
				[ '√((N+SX²÷SXX)÷N²)' '1÷√ SXX' ] × →Num
			»
		»
	sLR
		«
			ΣSize ΣX ΣY ΣXY ΣX² ΣY²
			→ N SX SY SXY SXX SYY
			«
				'√((SYY-Slope·SXY-Intercept·SY)·SXX÷(N-2)÷(N·SXX-SX²))'
				[ 1 '√(N÷SXX)' ] × →Num
			»
		»
	ΣParameters
		{ 1 2 1 ⁴⁹/₅ LinearFit }
	ΣData
		[[ 1 11 ]
		  [ 2 19 ]
		  [ 3 33 ]
		  [ 4 39 ]]
}
```

prismv, 2025-04-11:
A month ago I tried to create these program with direct usage of `ΣX²` and `ΣY²`, also without local variables at all, and it was not working - _therefore_ this issue as described above.
Now my 'strange' workaround (i.e. functioning a month ago) code is not working for `'ΣY²'`. So this is the remaining issue or regression.

"I don't see the point of quoting the values you pass to local variables." - I didn't see it either, but the 'normal' way was not working, but I wanted to have working programs to evaluate std.dev. of regression results.

One remaining question (at least for me) might be, if quoted commands should work at all, if non-quoted commands work.
But if so, it would be great to get the regression solved for the workaround to circumvent this issue.

prismv, 2025-04-12:
Thank you for the fix!

commit effa66ef (2025-04-11): parser: Fix parsing of `ΣY²` in equations

This is a bit special because `²` is also used for square

Fixes: #1426
closed 2025-04-12 by prismv