c3d/db48x #1427: Potential C++ compatibility problems on non Fedora building setups.

LdBeth, 2025-03-11:
For some reason, `make dm32` gives an error on my compilation setup (Void Linux on WSL2), while `make all` `make sim` both works.

The compilation error is:
```
/usr/lib/gcc/arm-none-eabi/9.3.0/../../../../arm-none-eabi/bin/ld: build/dm32/release/files.o: in function files::store_grob(text const*, grob const*) const [clone .part.0]':
files.cc:(.text._ZNK5files10store_grobEPK4textPK4grob.part.0+0x3d0): undefined reference to le<long>::byte_order'
/usr/lib/gcc/arm-none-eabi/9.3.0/../../../../arm-none-eabi/bin/ld: files.cc:(.text._ZNK5files10store_grobEPK4textPK4grob.part.0+0x3d4): undefined reference to le<unsigned long>::byte_order'
```

To fix that I haves to add `inline` keyword to `byte_order`, which leads to a 946kb `db50x.pg5`.
```
diff --git a/src/files.cc b/src/files.cc
index dddca62..a9a5ce2 100644
--- a/src/files.cc
+++ b/src/files.cc
@@ -703,7 +703,7 @@ struct le
 //   Little endian representation for a type
 // ----------------------------------------------------------------------------
 {
-    static constexpr union bytes
+    static inline constexpr union bytes
     {
         byte bytes[sizeof(Int)];
         Int  value;
```

Also, in `tools/forcecrc32/forcecrc32.c`, the GCC is not very happy about `FILE f[static 1]`, for the possible reason that `FILE` is an opaque type
definition in musl's `stdio.h` header. Replace it by `FILE *f` solves the problem.

prismv, 2025-03-11:
Could be that the compiler has to 'know' that `inline` has to be assumed implicitely, but I'm certainly not a C++ reference expert.
Afaik the hardware builds are done by "GNU Arm Embedded Toolchain 10.3-2021.10" - could be that the 9.3.0 is not the optimum choice.

I can confirm that one has to make sure that 

```
make tools/decimize/decimize tools/ttf2font/ttf2font tools/forcecrc32/forcecrc32
```

is compiled for the local host, therefore not by the embedded toolchain. This is not ensured by Makefile definitions for these targets.

LdBeth, 2025-03-11:
Right, that would explain the size difference in the compiled binary. I’ll see if I can setup the environment with toolchain downloaded from arm’s website 

And indeed the font tools are compiled by host compiler, which is the reason I brings about musl

LdBeth, 2025-03-12:
Since I cannot find a newer GCC Arm for Void Linux, I tried the setup on Apple Silicon Mac, which the oldest toolchain version can find is GCC Arm 12.2.Rel1, and the problem is gcc complains "build/dm32/small/db50x.elf section `.bss' will not fit in region `RAM'" and "region `RAM' overflowed by 1424 bytes", I looked at the memory map and find some additional C/C++ library functions are been pulled in. A bit of research bring to the conclusion that the new Arm GNU toolchain by default links to a more bloated newlib, compared to the old "Arm Embedded Toolchain".

The solution is to add `-specs=nano.specs -specs=nosys.specs` to build flags.

And as you said

> Could be that the compiler has to 'know' that inline has to be assumed implicitely,

The error related to `byte_order` is gone.

LdBeth, 2025-03-12:
So I found the text rendered in `SHOW` is dimmed, tried to copy the generated C++ files for font data from WSL but did not fix that. Unsure what can be the cause.

c3d, 2025-03-13:
Thanks for the investigation.

Indeed, I believe that gcc 9.3 is in error here. The presence of `inline` should definitely not change the result.

As for the binary size issue: I am really skimming very close to the limit of what can fit inside a real DM32 and DM42. This is the reason for a careful choice of specific compiler versions and options. Indeed, some other compilers cannot optimize sufficiently for the program to fit in the available hardware. To this, unfortunately, I have not better answer than the one you found, which is to tweak options for your compiler until you can make it fit.

Thanks for documenting the `-specs=nano.specs -specs=nosys.specs` option. That may prove useful one day if I'm forced to change compilers.

c3d, 2025-03-13:
@LdBeth, I hope that you won't mind if I mark this as `wontfix` and close the issue. We cannot support older compilers like GCC 9.3 if they have bugs that have been fixed later. If you submit a patch with your `-spec` changes, I can accept it if it does not negatively impact my own builds. Otherwise, I will modify it to make it conditional on the compiler.

For now, closing this issue, please open another issue referencing this one if you plan to submit a change with the `-spec` options.

Thanks for your detailed analysis.

closed 2025-03-13 by c3d