c3d/db48x #1429: Bug in the function SigDig

Wiljea, 2025-03-13:
I discover the following bug about the function SigDig to compute the number of significant digits for number greater than "0" having trailing "0":
7 SigDig = 1 OK 
70 SigDig = 1 OK
700 SigDig = 1 OK
7000 SigDig = 3 NOT OK, should be 1 
70000 SigDig = 3 NOT OK, should be 1 
700000 SigDig = 3 NOT OK, should still be 1
...

c3d, 2025-03-13:
Good catch, @WilJea. This issue only happens when an integer to decimal conversion happens. In other words, `7000. SigDig` returns `1` as intended.

There is a bug in that integer to decimal conversion, which can be reduced to the following case:

```
7000 TODECIMAL BYTES
@ Produces #AF02 0428h 6

7000. BYTES
@ Produces #AF01 0428h 5
```

In the first case, we produce a decimal number where the trailing zeroes are not correctly stripped. The internal representation otherwise remains correct, so the rest of the computations will not be affected, but it throws `SigDig` off.

Wiljea, 2025-03-14:
But it seems that, in some case, SigDig doesn't produce the right result even with a decimal number:
70000. SigDig => 1 OK
70000 →NUM SigDig => 3 NOT OK even if after →NUM, we have 70000. on the stack.

Wiljea, 2025-03-26:
Done.

commit c1bb86a1 (2025-03-14): decimal: Strip trailing zeros from integers

Converting integer values ot decimal should strip the trailing zeros,
to avoid creating non-normalized numbers.

Fixes: #1429
commit 08b4dd04 (2025-03-15): decimal: Strip trailing zeros from integers

Converting integer values ot decimal should strip the trailing zeros,
to avoid creating non-normalized numbers.

Fixes: #1429
commit 66df1393 (2025-04-06): decimal: Strip trailing zeros from integers

Converting integer values ot decimal should strip the trailing zeros,
to avoid creating non-normalized numbers.

Fixes: #1429
closed 2025-03-26 by Wiljea