c3d/db48x #1430: Makefile change to support newer arm compilers

prismv, 2025-03-13:
Please change the Makefile according to the proposed patch of @LdBeth in #1427, because it allows the usage of current versions of arm cross-compilers :

`-specs=nano.specs -specs=nosys.specs` added to `PLATFORM_FLAGS` and `LINK_OPTS`. 

An alternative proposal might be to allow external definitions by setting these variables only with `+=` within Makefile to allow addition of such setting on `make` call.

Compilation works then e.g. with:

```
❯ arm-none-eabi-c++ --version
arm-none-eabi-c++ (Arch Repository) 14.2.0
Copyright (C) 2024 Free Software Foundation, Inc.
```

prismv, 2025-03-13:
Checked elf size information (and pgm size explicitely, too) of 3 variants based on 0.9.2 code:

##### unmodified code, arm-none-eabi-g++ (GNU Arm Embedded Toolchain 10.3-2021.10) 10.3.1 20210824 (release)

``` shell
❯ arm-none-eabi-size -A build/dm42/release/db48x.elf
build/dm42/release/db48x.elf  :
section             size         addr
.qspi             279296   2415919104
.rodata            84280    134545408
.text             568320    134629696
.ARM.extab             0    135198016
.ARM                   8    135198016
.preinit_array         0    135198024
.init_array           24    135198024
.fini_array            8    135198048
.data               4584    268435456
.bss                3432    268440040
.ARM.attributes       48            0
.comment              73            0
.debug_frame        7756            0
Total             947829
```

pgm file size is 657252.

##### Makefile with nano and nosys specs, arm-none-eabi-g++ (GNU Arm Embedded Toolchain 10.3-2021.10) 10.3.1 20210824 (release)

``` shell
❯ arm-none-eabi-size -A build/dm42/release/db48x.elf
build/dm42/release/db48x.elf  :
section             size         addr
.qspi             279296   2415919104
.rodata            83872    134545408
.text             542160    134629280
.ARM.extab             0    135171440
.ARM                   8    135171440
.preinit_array         0    135171448
.init_array           16    135171448
.fini_array            8    135171464
.data               3608    268435456
.bss                3424    268439064
.ARM.attributes       48            0
.comment              73            0
.debug_frame        6056            0
Total             918569
```

pgm file size is 629692 bytes.

###### Makefile with nano and nosys specs, arm-none-eabi-g++ (Arch Repository) 14.2.0

``` shell
❯ arm-none-eabi-size -A build/dm42/release/db48x.elf
build/dm42/release/db48x.elf  :
section              size         addr
.qspi              279280   2415919104
.rodata             83688    134545408
.text              547848    134629104
.ARM.extab              0    135176952
.ARM                    8    135176952
.preinit_array          0    135176960
.init_array            16    135176960
.fini_array             8    135176976
.data                3592    268435456
.bss                 4136    268439048
.ARM.attributes        48            0
.comment               30            0
.debug_line         49788            0
.debug_line_str       378            0
.debug_info        117942            0
.debug_abbrev       28855            0
.debug_aranges       3248            0
.debug_str          12867            0
.debug_frame         6156            0
.debug_loclists     33638            0
.debug_rnglists      2464            0
Total             1173990
```

pgm file size is 635188 bytes.

prismv, 2025-03-20:
Because I still use the pgm compiled with 14.2 cross-compiler, I can confirm an issue that the exponent is not displayed.

Still have to confirm with 'standard' version, if this is an compiler issue - or an C++ language version issue? - but there I would expect same normal behaviour as seen in simulator.

LdBeth, 2025-03-20:
Thanks.

The problem appears to be the fancy branch does not work as intended, I think the problem is with the usage of snprintf function, maybe the nosys profile only provides a stub implementation thus nothing is printed.

https://github.com/c3d/db48x/blob/a5876494975bca2511896334540a2358277c82a6/src/decimal.cc#L660

LdBeth, 2025-03-20:
I did some experiment and reaches the conclusion that newlib nano does not seem support `"%lld"` or `"%" PRId64`, "%d" would work.

Searching on web does confirm that newlib nano now default does not support long long int. The solution would be rebuild the newlib nano shipped with the arm gcc toolchain with appropriate options.

prismv, 2025-03-20:
Great findings.

> I did some experiment and reaches the conclusion that newlib nano does not seem support `"%lld"` or `"%" PRId64`, only `"%d"` would work.
> 
> Searching on web does confirm that newlib nano now default does not support long long int. The proper solution would be rebuild the newlib with appropriate custom options so it can both support long long int type and still fit into dm42’s ram.
> 

Could be that the newlib-nano does not support long long int in printf at all - so not sure about feasibillity of 'custom options' .

> However, since the non-fancy version used by SHOW command uses `"%d"`, I guess we can just cast the exponent from long long int to `int`. And by doing that I did get the exponent display to work again.

I'll try to add this to the merge request. Should be acceptable, if the usage "%d" for ClassicExponent view is not seen as a new bug ;-)

prismv, 2025-03-20:
[83afedc](https://github.com/c3d/db48x/pull/1431/commits/83afedcd287b12fbf9810b685940d25535738936) - hopefully same output for fancy and classic exponent by usage of `PRId32` in `snprintf` and `"%ld"` in `r.printf`.
Format strings might be further unified, if such a change is in principle accepted.

prismv, 2025-03-20:
@LdBeth After correction [4ce11a5](https://github.com/c3d/db48x/pull/1431/commits/4ce11a500d35cb2f4b8f0d16e964fefdde4a4251) I realize that the implementation is such that with FancyExponent the variable precision implementation is supported with max of 64 bit Integers.

So 1.⁳²¹⁴⁷⁴⁸³⁶⁴⁸ is supported in fancy mode, but not in classic mode - this might be the desired behaviour @c3d ?

So this might be the best way to support experiments with other compilers and settings:

> 
> An alternative proposal might be to allow external definitions by setting these variables only with `+=` within Makefile to allow addition of such setting on `make` call.
>

LdBeth, 2025-03-20:
> Could be that the newlib-nano does not support long long int in printf at all

From various descriptions online it seems that is the case. However, build a trimmed down version from newlib should still be viable.

I suggest before making further changes, let's first settle:
1. should we restrict usage of `%lld` in the future developments?
2. is 32-bit integer good enough for exponents?

prismv, 2025-03-21:
> From various descriptions online it seems that is the case. However, build a trimmed down version from newlib should still be viable.
>
> I suggest before making further changes, let's first settle: [...]

I could live with 32-bit integers for exponents for all practical applications that come to my mind.

It's also worth to mention this clearly related issue: Although `%lld` is currently supported for FancyExponent, only `%ld` cast of the exponent is saved in state file.

LdBeth, 2025-04-07:
Then when I try to play with hardware fp (also needs to set precision to 16), I realized it is also rely on printf to format machine float. This means `-specs=nano.specs` needs to be `-specs=nano.specs -u _printf_float` to make machine float display function work properly. Fortunately this does not seem increase the size of the binary.

prismv, 2025-04-07:
I simply trusted @LdBeth report here and added option in [4f15aae](https://github.com/prismv/db48x/commit/4f15aae40e69cb03a2928d2c2beac0721c1a3f6b). Short test gives exponent with HWFP enabled and Prec 16 setting.

c3d, 2025-04-10:
> [@LdBeth](https://github.com/LdBeth) After correction [4ce11a5](https://github.com/c3d/db48x/pull/1431/commits/4ce11a500d35cb2f4b8f0d16e964fefdde4a4251) I realize that the implementation is such [...]

Finally got back to being able to look at this issue and the related PR.

No, @LdBeth, truncating large exponents in non-fancy mode is not intended. So good find about non-fancy exponent not displaying correctly when the exponent is large. I just fixed it with `PRId64` on the non-fancy side.

Did you find a way to get %lld support / 64-bit printing working with the `nano` spec?

> 
> So this might be the best way to support experiments with other compilers and settings:
> 
> > An alternative proposal might be to allow external definitions by setting these variables only with `+=` within Makefile to allow addition of such setting on `make` call.

Yes, this could help.

LdBeth, 2025-04-10:
> truncating large exponents in non-fancy mode is not intended. So good find about non-fancy exponent not displaying correctly when the exponent is large. I just fixed it with `PRId64` on the non-fancy side.

fwd to @prismv :)

> Did you find a way to get %lld support / 64-bit printing working with the `nano` spec?

The conclusion is that it is impossible to enable `%lld` for nano spec without recompile the installed newlib.

If 64-bit printing is desired, I'll try if building the custom newlib can work.

c3d, 2025-04-10:
Thanks for this research. I pushed on `dev` a squash commit of a reduced version of your changes, with additional code to correctly render 64-bit exponents without using the unsupported `PRId64` format. Please give it a try and if that works for you we can close this issue and the PR.

LdBeth, 2025-04-10:
Even with the custom newlib been built, the libc size still looks chunky and I cannot figure out a proper way to drop in replace the default newlib unless rebuild the entire GCC toolchain. So I'd give up on custom newlib.

>  I pushed on dev a squash commit of a reduced version of your changes, with additional code to correctly render 64-bit exponents without using the unsupported PRId64 format. Please give it a try and if that works for you we can close this issue and the PR.

Installed on my DM42n and looks like it works fine. With -O3 the sizes are:

```
-rw-r--r--    1 ldbeth  staff   633K Apr 10 18:33 db48x.pgm
-rwxr-xr-x    1 ldbeth  staff   278K Apr 10 18:26 db50x_qspi.bin
-rw-r--r--    1 ldbeth  staff   992K Apr 10 18:24 db50x.pg5
```

prismv, 2025-04-11:
@c3d your pointer arithmetics instead of snprintf work fine. It works for large exponents also when compiled with gcc 14.2.

So thanks to the proposal of @LdBeth  additionally to decreased program sizes it is also easier to test compilations with newer versions of arm cross-compilers. 

Note that issue #1286 is still relevant for lots of program examples that give error 'Malformed local program', if gcc 14.2 cross-compiler is used. With this change it will be easier to debug (sometime later), why ...

closed 2025-04-11 by prismv