c3d/db48x #1434: Mixing algebraic and reverse-polish operations does not work as documented.

JohnHind, 2025-03-17:
The documentation file gives the following example:
`A second method is to mix and match algebraic and RPL, by typing the following sequence: '(), √x, 2, x², +, 3, x², ENTER. At this point, the expression without the multiplication is on the stack. We can then multiply it by 1000 by typing 1, 0, 0, 0, ×. The expression with the multiplication is then shown on the stack, and can then be evaluated with the = key.`
This does not give an expression when the multiplication key is typed but rather evaluates a numerical result immediately without waiting for the = key.
I have observed this behaviour on the browser version and on the iOS app on my iPad. I also tried reseting the state on the iPad app to make sure I'd not set any odd preferences.
It is a pity this does not work as documented, because it seems like a really nice feature!

LdBeth, 2025-03-17:
You will need to the Simplification option to false.

It is in MODES > Math > Simpl

JohnHind, 2025-03-17:
It seems inconsistent that the `AutoSimplify` option simplifies an expression after it is extended, but not on the initial ENTER. But in either case I'd definitely prefer for it to be simplified only when I specifically request it (after I've finished composing it). The documentation definitely needs to mention if an example depends on a non-default option setting.

LdBeth, 2025-03-18:
ENTER never evaluates the input, which is the intended behavior.

c3d, 2025-03-20:
I think that I agree with @JohnHind that this is not intended behaviour. It is a regression, and it definitely does not happen on the HP50G.

@LdBeth, thanks for pointing out the role of `AutoSimplify`, but that should not really matter here IMHO.

commit 5124c9bd (2025-04-26): arithmetic: Avoid premature simplification during operations

When doing `'sqrt(sq(2)+sq(3))' 1000 *` we would simplify the result,
which would cause evaluation of the constant on the left, instead of
producing the expected combined expression.

This is inconsistent with both what the HP50G does and what is
documented in section _Mixing algebraic and reverse-polish operations_
of the documenation [1].

[1]: https://github.com/c3d/db48x/blob/stable/doc/2-Quickstart.md#mixing-algebraic-and-reverse-polish-operations

Fixes: #1434
closed 2025-04-27 by c3d