c3d/db48x #1435: Suggestion about reducing fractions to decimals

JohnHind, 2025-03-17:
Enter a fraction: 45 ENTER 2365 S and the stack shows a fraction 9/473. Hit the O key and this is reduced to the decimal value 0.019...
Nothing on the O key label is mnemonic for this behaviour, and the key labelled = seems a much more natural choice for this function. At present it does nothing with a fraction on the stack, so could be reassigned to this. Reducing a fraction to a decimal value is intuitively the same as evaluating an expression (to me).
Furthermore, the current assignment to the O key _does_ have a conflict. If the fraction has units assigned to it, for example 9/473mm, the O key cycles through the SI prefixes leaving the value as a fraction and there seems to be no way to reduce it to a decimal. Assigning the decimal reduction to the = key and leaving the cycle function on the O key would solve this problem as well as (IMHO) being more intuitive.

LdBeth, 2025-03-17:
What you want is probably `→NUM`, which already binds to Shift 1 and is the usual command used to convert a fraction to decimal on HP49 series, the EVAL command never changes mode of the number on the stack regardless of current mode.

JohnHind, 2025-03-17:
I'm using the browser simulator and the iPad app. On the 1 key left-shift function, the former shows ASN until you click into it whereupon it changes to ->NUM, the iPad app shows ASN always. In both cases it does reduce a fraction to a decimal. The documentation however does not mention this and says to use the O key, which only works if the fraction has no unit with a cycle associated (but it does toggle between decimal and fraction without needing another key to go back to fraction). I still think the = key would be a more intuitive assignment.

c3d, 2025-04-03:
Hi @JohnHind. Thank you for your suggestion, but the short answer is no.

The long-form rationale for this answer is given in [this video](https://www.youtube.com/watch?v=jNOA39HnkcM&list=PLz1qkflzABy-Cs1R07zGB8A9K5Yjolmlf&index=30), and specifically in the section named "_Exact computations_".

I'll try to summarize it here: in order to give good results, DB48x tries hard
to perform exact computations for as long as you can. If that's not what you
want, then you can use the `NumericalResults` setting instead of the default
`SymbolicResults`. With `NumericalResults`, `2 3 /` will give you `0.66666 66666 67`. This command is in the `MathModesMenu`, Shift-N, F4, F1.

The `=` key maps to `Run`, which evaluates what is on level 1 of the stack, and differs from `Eval` in that it also resumes execution if in a `Prompt`. This is an extremely important feature, with a fundamental property: it is a one-way street. In other words, when you evaluate `'2*3+5'` with the `=` key, it does not go back.

By contrast, the `O` key maps to the `Cycle` command, which as its name indicates, is cyclical. It is a change of format, and it tries to not be too destructive. For example, for a fraction, it _cycles_ back and forth between various representations of the number.

With that context in mind, let me now address your position. You wrote:

> Nothing on the O key label is mnemonic for this behaviour

There are currently three possible labels for this key: E, EEX and x10ⁿ.
During data entry, the key is traditionally intended to enter an exponent, but it now has additional uses, notably cycling orders of magnitude prefixes for units, e.g. if the command line contains `1_m` and you hit that key, you get `1_km`, and if you do it again, you get `1_cm`. Both entering exponents and this unit size cycling are, I hope, obviously connected to the x10ⁿ label (or the historical E and EEX labels).

If you are not entering a value, being able to do a quick cycle between common related presentations is a feature that I wanted to be on a quick-access key, i.e. a non-shifted one. There aren't that many left on the keyboard, but to me associaating it with x10ⁿ makes perfect sense:

- For units, it changes the prefix during entry, so it's essentially changing the power of ten in a value-preserving way

- For fractions, it toggles between a decimal representation (which I myself associate easily with the label x10ⁿ) and a fraction representation

- For units, it turns out that there was a request to have configurable unit cycles, which are defined in the `conifg/units.csv` file. One of my early users wanted quick conversions betweemn `in` and `mm`, so I left it there. This is why `mm` behaves differently from other units. I should probably remove this example, since there is another useful example with monetery units that does the trick. Anyway, if you try with, say, `1.2_A`, you will be able to check that the `Cycle` command will adjust the order of magnitude, again tightly related to the x10ⁿ label, preserving the value, and cycling betwen decimal and fractional representation.

- Finally, for me at least, the letter `O`, which is round, is a reminder that this is cyclical. So I have at least another mnemonic way to remember what this is about.

> Reducing a fraction to a decimal value is intuitively the same as evaluating an expression (to me)

Maybe to you, not to me. One operation is one way, the other is cyclical. Entering `1.2` then hitting `O` will give me `⁶/₅`, which is quite useful too.

Also, as others have pointed out, converting to decimal also has an easy access dedicated key (shift 1). `ASN` is the historical label (for DM42-compatibility), and I reinterpreted that as "As Number" (on DM42, it is "Assign"). It's a bit difficult to find key mappings on this cramped keyboard, even more so some that are vaguely connected to the DM42 key layout. Nothing will be perfect.

In summary, `Cycle` is one of the enhancements in DB48x over classical RPL that I find the most incredibly useful. Don't confuse it with `Eval` or `Run` and you will see how incredibly convenient it is in practice.

closed 2025-04-03 by c3d

JohnHind, 2025-04-03:
Hi Christophe, sorry to reopen this after you gave such a detailed and helpful explanation.

I think I have understood that, on the stack, an expression and a fraction are different things even though they look the same (except for a slightly different font size). Maybe the visual representation could be more distinct? How about offsetting the numerator/denominator by half a character and tilting the separator for fractions?

Given that the fraction is an exact representation which may be displayed as a (possibly approximate) decimal, units could be treated in the same way (the underlying representation remains as entered, but a conversion factor is applied for display). At the moment, applying the cycle key to a fraction with units results in decidedly odd, non-cyclical behaviour! Place 1/2 (a fraction) on the stack and assign units of metres to it from the UNITS menu. Observe what happens when you press the cycle key repeatedly: whatever it is it certainly is not a cycle! Ultimately it ends up cycling round a short list of in,mm,cm never returning to the original metres or the original fractional representation!

Maybe also there could be a setting for imperial/metric user preference. Then the cycle would first convert to the preferred system (if necessary) and then cycle through a separate list for each system. So for metric preference, inches would first convert to cm and then cycle round the metric prefixes. For imperial preference, cm would convert to inches and then cycle through maybe feet, yards, miles and back to inches.