c3d/db48x #1436: Suggestion for "full stack" manipulation: Stk↑, Stk↓, Stk↑N & Stk↓N

Wiljea, 2025-03-19:
Since these days I'm playing with programs producing more than 100 entries to the stack, it would be interesting and quite useful to have "full stack" manipulation to rotate (1 or N steps) the entire stack as a whole (like the action of R↑ & R↓ to the 4-level stack of RPN calculator):

**StackUp (Stk↑)** 
For a 1 level move of the whole stack upward it would execute:
```rpl
« Depth Roll »
```

StackDown (Stk↓)
For a 1 level move of the whole stack downward it would execute: 
```rpl
« Depth RollDown » 
```

**StackUpN (Stk↑N)** 
For a N level move of the whole stack upward it would execute with N as input:
```rpl
« 1 Swap For j Depth Roll Next »
```
 
StackDownN (Stk↓N)
For a N level move of the whole stack downward it would execute with N as input: 
```rpl
« 1 Swap For j Depth RollDown Next »
```

What do you think of this ? The second page of the menu could be (suggestion):
|  Undo   | 
| NDupN | DupDup |  ClrStk   |    Stk↓   |   Stk↓N |
|  Swap    |  LastArg |  LastX    |    Stk↑   |   Stk↑N |

LdBeth, 2025-04-01:
For more complicated programs I'd much prefer to write it in C++ than UserRPL. Hope this can be soon covered in the tutorials.

Wiljea, 2025-04-02:
I am not sure anymore of the pertinence of these 4 functions that I suggested, since today I played a lot with the various functions of the **Interactive Stack**. In summary they do the same if not better. By the way, I am surprised to see how the **Interactive Stack** is poorly documented, here and also in the reference manuals of either HP48 & HP50g. Maybe I didn't look at the right place !

I think, I will soon close this ISSUE as being irrelevant.

c3d, 2025-04-02:
These commands seem simple enough that they could make nice additions to the RPL library.

c3d, 2025-04-02:
> @LdBeth For more complicated programs I'd much prefer to write it in C++ than
> UserRPL. Hope this can be soon covered in the tutorials.

Are you asking about documenting the internals of DB48x, a sort of developer's guide? IF so, that does not exist yet, although several YouTube videos give some [design constraints and implementation details](https://www.youtube.com/watch?v=jNOA39HnkcM&list=PLz1qkflzABy-Cs1R07zGB8A9K5Yjolmlf&index=30).

Or are you asking about an hypothetical (if vaguely planned) C++ SDK that would allow to build binary extensions to the core RPL engine? If so, "some day in a long long time" is probably the correct answer.

LdBeth, 2025-04-02:
@c3d a developer’s guide would suffice if that would allows adding a few additional RPL primitives by tweaking the source.

I guessed it’s probably not as easy as on SysRPL to call UserRPL words in C++.

c3d, 2025-04-28:
> [@c3d](https://github.com/c3d) a developer’s guide would suffice if that would allows adding a few additional RPL primitives by tweaking the source.

To add a primitive command, let's see what is needed for `TimedEval`:
1. Add [its spelling(s) in ids.tbl](https://github.com/c3d/db48x/blob/02409d2f698e28fd62b3d9f49b06c8fb1f2fd04e/src/ids.tbl#L687). `TimedEval` has an alias called `TEval`. The last alias is the one used in HP48 compatibility spellings, and its capitalization will be used in mixed case mode.
2. Add a declaration for the [command  in a header](https://github.com/c3d/db48x/blob/02409d2f698e28fd62b3d9f49b06c8fb1f2fd04e/src/datetime.h#L85), using the `COMMAND_DECLARE` macro. The second argument is the arity of the command (i.e. how many arguments it takes from the stack). 
3. Add an implementation for the [command in the corresponding C++ implementation file](https://github.com/c3d/db48x/blob/02409d2f698e28fd62b3d9f49b06c8fb1f2fd04e/src/datetime.cc#L938), using the `COMMAND_BODY` macro.

Functions (which you can use in algebraics) have different macros.

> I guessed it’s probably not as easy as on SysRPL to call UserRPL words in C++.

I picked up `TimedEval` because it [invokes `Evaluate` undearnath](https://github.com/c3d/db48x/blob/02409d2f698e28fd62b3d9f49b06c8fb1f2fd04e/src/datetime.cc#L944).

There is also an `evaluate_function` helper if you want to evaluate something that returns a value, e.g. [in plot](https://github.com/c3d/db48x/blob/02409d2f698e28fd62b3d9f49b06c8fb1f2fd04e/src/plot.cc#L261).

One subtlety about calling user code is that you need to create an object that [adjusts settings as necessary](https://github.com/c3d/db48x/blob/02409d2f698e28fd62b3d9f49b06c8fb1f2fd04e/src/plot.cc#L244) for whatever you evaluate. This is mostly to ensure we only save last arguents / undo stack on entry to *your* command (unlike the HP50G where Undo in a plot or solve is useless).

c3d, 2025-04-28:
Closing this specific issue, although writing a dev guide is definitely a good idea ;-)