c3d/db48x #1438: Unable to make the logical structure "CASE THEN END" to work properly

Wiljea, 2025-03-20:
The example provided in ISSUE #1437 used nested logical structures "IF THEN ELSE END" which is a typical situation to use the "CASE THEN END" structure. I did this because I was not able to have the latter to work properly.

The failed case:
```rpl
« Swap List→ Drop Duplicate2 if > then Swap end → n X1 X2 « CASE n 0 = THEN 1 1 END CASE n 2 mod 1 = X1 0 > or THEN X1 n ↑ X2 n ↑ END CASE X2 0 < n 2 mod 0 = and THEN X2 n ↑ X1 n ↑ END 0 X1 n ↑ X2 n ↑ Max Duplicate2 if > then Swap END 2 →List » »
```rpl

The working case: 
```rpl
« Swap List→ Drop Duplicate2 if > then Swap end → n X1 X2 « if n 0 = then 1 1 else if n 2 mod 1 = X1 0 > or then X1 n ↑ X2 n ↑ else if X2 0 < n 2 mod 0 = and then X2 n ↑ X1 n ↑ end end 0 X1 n ↑ X2 n ↑ Max Duplicate2 if > then Swap end end 2 →List » »
```
with the same examples as of ISSUE #1437. I don't know here also if me RPL use is in fault.

LdBeth, 2025-03-24:
The correct syntax should be
```
<<
  CASE
  test1 THEN action1 END
  test2 THEN action2 END
  ...
  testN THEN actionN END
  defaultaction
  END
>>
```

Wiljea, 2025-03-24:
@LdBeth 
Thanks, you were right in the sense that there was a missing "END" terminating the CASE command. However, if  I add it, it still doesn't work:
```rpl
« Swap List→ Drop Duplicate2 
if > 
     then Swap 
end → n X1 X2 
    « 
        CASE n 0 = 
             THEN 1 1 END 
        CASE n 2 mod 1 = X1 0 > or 
            THEN X1 n ↑ X2 n ↑ END 
        CASE X2 0 < n 2 mod 0 = and 
            THEN X2 n ↑ X1 n ↑ END 
        END
        0 X1 n ↑ X2 n ↑ Max Duplicate2 
        if > 
            then Swap 
        END 
        2 →List 
    » 
»
```
So, where is my remaining problem? Is it me or db48x which is in fault here?

LdBeth, 2025-03-24:
> No, surprisingly it still doesn't work for me. Again is it my RPL which is in error or a possible bug in db48x 

First, I'd suggest assume $x1 \leq x2$ and eliminate both of the `if > then swap end` part.

Second, in `0 X1 n ↑ X2 n ↑ Max Duplicate2` the `Duplicate2` word should be removed.

After these two changes the program would implement the intended algorithm.

Wiljea, 2025-03-24:
Thanks a lot ! I maintained "Swap List→ Drop Duplicate2 if > then Swap end" at the beginning instead of "Swap List→ Drop" to correct for the order in the input interval if necessary and now it works fine:
ex.1: { -2 4 } 0 IntToN => { 1 1 }
ex.2: { -2 -4 } 3 IntToN => { -64 -8 }
ex.3: { -2 -4 } 4 IntToN => { 16 256 }
ex.4:   { -3 4 } 3 IntToN => { -27 64 }
ex.5:   { -1 2 } -3 IntToN => { -1 ¹/₈ }
ex.6:   { -2 4 } -4 IntToN => { 0 ¹/₁₆ }
ex.7:   { -2 -5 } -3 IntToN => { -¹/₁₂₅ -¹/₈ }
ex.8:   { -2 0 } 2 IntToN => { 0 4 }
ex.9:   { -3 0 } 3 IntToN => { -27 0 }
ISSUE to be closed.

closed 2025-03-24 by Wiljea