c3d/db48x #1439: Help required to identify a rare random bug in a program testing for concordance between two intervals

Wiljea, 2025-03-26:
The following program tests for concordance between two input intervals of the form IY = {Y1 Y2} & IX = {X1 X2} and illustrates the intersection IY ∩ IX = {Z1 Z2} if it is not empty and gives diagnostics (otherwise or with % P1 & % P2 of coverture) (Note: Xi, Yi & Zi may have units and are not a priori ordered)
```rpl
«
	Swap List→ Drop Duplicate2
	if
		>
	then
		Swap
	end
	2 →List Swap List→
	Drop Duplicate2
	if
		>
	then
		Swap
	end
	2 →List Swap List→
	Drop Rot List→ Drop
	0 0 0 0 0 0 0
	→ Y1 Y2 X1 X2 Z1 Z2
	P1 P2 XMIN XMAX XDELTA
	«
		X1 Y1 Max 'Z1' Store
		X2 Y2 Min 'Z2' Store @ 
		X1 Y1 Min 'XMIN' Store
		X2 Y2 Max 'XMAX' Store
		XMAX XMIN - 'XDELTA' @ XMIN, XMAX & XDELTA control the PlotXRange
		Store XMIN XDELTA 20
		÷ - 'XMIN' Store XMAX
		XDELTA 20 ÷ + 'XMAX'
		Store XMIN UnitValue
		XMAX UnitValue PlotXRange
		-0.4 10 PlotYRange @ -0.4 & 10 control the PlotYRange and are fixed for all trials
		"Concordance TEST Illustration"
		XMIN 0.08 XDELTA ×
		+ UnitValue 10 ℝ→ℂ
		DrawText X1 UnitValue
		7.8 ℝ→ℂ X2 UnitValue
		7.8 ℝ→ℂ DrawLine
		"|" X1 UnitValue 8.5
		ℝ→ℂ DrawText "|"
		X2 UnitValue 8.5 ℝ→ℂ
		DrawText Y1 UnitValue
		6.2 ℝ→ℂ Y2 UnitValue
		6.2 ℝ→ℂ DrawLine
		"|" Y1 UnitValue 7 ℝ→ℂ
		DrawText "|" Y2 UnitValue
		7 ℝ→ℂ DrawText
		if
			Z1 Z2 =
		then
			Z1 UnitValue 4.7 ℝ→ℂ
			Z2 UnitValue 4.7 ℝ→ℂ
			DrawLine "|" Z1 UnitValue
			5.5 ℝ→ℂ DrawText
			"IY ∩ IX = {" Z1
			+ " " + Z2 + "} " +
			XMIN UnitValue 4 ℝ→ℂ
			DrawText "Concordance in one point only of"
			XMIN 0.01 XDELTA
			× + UnitValue 2.5
			ℝ→ℂ DrawText
			"IX={" X1 + " " + X2
			+ "}; IY={" + Y1 +
			" " + Y2 + "}" + XMIN
			0.01 XDELTA × +
			UnitValue 1 ℝ→ℂ
			DrawText
		end
		if
			Z1 Z2 <
		then
			Z1 UnitValue 4.7 ℝ→ℂ
			Z2 UnitValue 4.7 ℝ→ℂ
			DrawLine "|" Z1 UnitValue
			5.5 ℝ→ℂ DrawText
			"|" Z2 UnitValue 5.5
			ℝ→ℂ DrawText
			"IY ∩ IX = {" Z1
			+ " " + Z2 + "} =>"
			+ XMIN 0.04 XDELTA
			× + UnitValue 4 ℝ→ℂ
			DrawText
			if
				X1 Z1 ≥
			then
				Z2 Z1 X1 Max - X2
				X1 - ÷ 100 × →Num
				2 Negate Round 'P1'
				Store Z2 Y2 Min Z1
				- Y2 Y1 - ÷ 100 ×
				→Num 2 Negate Round
				'P2' Store P1 "% of IX = {"
				+ X1 + " " + X2 +
				"};" + XMIN 0.08
				XDELTA × + UnitValue
				2.5 ℝ→ℂ DrawText
				P2 "% of IY = {" +
				Y1 + " " + Y2 + "}"
				+ XMIN 0.08 XDELTA
				× + UnitValue 1 ℝ→ℂ
				DrawText
			end
			if
				Y1 Z1 ≥
			then
				Z2 X2 Min Z1 - X2
				X1 - ÷ 100 × →Num
				2 Negate Round 'P1'
				Store Z2 Z1 Y1 Max
				- Y2 Y1 - ÷ 100 ×
				→Num 2 Negate Round
				'P2' Store P1 "% of IX = {"
				+ X1 + " " + X2 +
				"};" + XMIN 0.08
				XDELTA × + UnitValue
				2.5 ℝ→ℂ DrawText
				P2 "% of IY = {" +
				Y1 + " " + Y2 + "}"
				+ XMIN 0.08 XDELTA
				× + UnitValue 1 ℝ→ℂ
				DrawText
			end
		else
			if
				Z1 Z2 >
			then
				"IY ∩ IX is empty "
				XMIN 0.04 XDELTA
				× + UnitValue 5.5
				ℝ→ℂ DrawText
				"=> No concordance for"
				XMIN 0.08 XDELTA
				× + UnitValue 4 ℝ→ℂ
				DrawText "IX = {"
				X1 + " " + X2 + "};"
				+ XMIN 0.08 XDELTA
				× + UnitValue 2.5
				ℝ→ℂ DrawText
				"IY = {" Y1 + " "
				+ Y2 + "}" + XMIN
				0.08 XDELTA × +
				UnitValue 1 ℝ→ℂ
				DrawText
			end
		end
	»
»
```

The program is named IntTest. I submit IntTest to a Monte Carlo type of test using as input a few hundreds of random inputs formed by couple of intervals (see below for the program IntRDM & its subroutine to produce these random inputs with limits chosen between -100 & 100). For 200 couples of intervals, it appears that IntTest fails 3% of them for the following 6 interval couples:
{43 88} {57 84} ; { 22 36 } { 64 86 } ; { 26 74 } { 15 45 } ; { 34 37 } { 58 59 } ; { 16 48 } { 42 93 } ; { 35 71 } { 66 79 }   [1] 

When IntTest fails I obtain the message "PlotYRange error : Invalid plot parameters" and the program stops at the instruction "-0.4 10 PlotYRange" (whose parameters are fixed and do not change for all trials). Then if you try immediately another interval couple, it fails too with "DrawLine error: Invalid plot perameters", but this is not a true failure because if you retry it it will pass without error. This might point to a possible memory corruption cause, but I'm really not sure of what is happening. When everything works fine, you will obtain one of the following 4 cases:

**Examples of IntTest**:
ex.1: { -8_m 23_m } { 23_m 42_m } IntTest
[image]

ex.2: { 52_s 83_s } { -8_s 61_s } IntTest
[image]

ex.3: { -19_m/s 20_m/s} {10_m/s 45_m/s} IntTest
[image]

ex.4: {20_km/h 70_km/h} {10_km/h -40_km/h} IntTest
[image]

**Program to generate random couple of intervals**:

The main program IntRDM takes a number N as input. If "N≠1" it gives a couple of truly (concordant or not) random intervals (with limits between -100 and 100), if "N=1" it gives a couple of concordant random intervals within the same limits:
```rpl
«
	→ N
	«
		0 0 0 0 0 0
		do
			6 DropN IntToss
		until
			if
				N 1 =
			then
				Duplicate2 ≤
			else
				1
			end
		end
		Drop Drop 2 →List
		UnRot 2 →List Swap
	»
»
```
The following subprogram IntToss calculates the limits of intervals Y1 Y2 X1 X2 Z1 Z2 of Intervals IY, IX and the possible intersection IZ:
```rpl
«
	«
		«
			100 Duplicate Negate
			Swap Random
		»
		→ XToss @ a local subprogram to sample one interval limit A between -100 & 100
		«
			XToss XToss Duplicate2
			if
				>
			then
				Swap
			end
			XToss XToss Duplicate2
			if
				>
			then
				Swap
			end
		»
	»
	→ IYIXToss @ a local subprogram producing two ordered intervals Y1, Y2 & X1, X2
	«
		IYIXToss 0 0 0 0
		→ Y1 Y2 X1 X2 Z1 Z2
		IY IX
		«
			X1 Y1 Max 'Z1' Store  @ Main program calculating Z1 & Z2 of IZ
			X2 Y2 Min 'Z2' Store
			Y1 Y2 X1 X2 Z1 Z2
		»
	»
»
```

Wiljea, 2025-03-26:
@LdBeth  Due to your competence in RPL, I would appreciate your opinion or comments if you have aa look at the program in issue #1439. Thanks in advance. This also apply to other readers: your advice will be welcome.

Wiljea, 2025-03-29:
Looking at which variable is directly linked to the cause of the bug, one can observe that the program behaves as expected for the couple of intervals {43 88} {13 84} but fails for the next couple {43 88} {14 84} where only X1 was increased by one unity since the input IY IX goes to the variables: {Y1 Y2} {X1 X2}. Clearly the value of X1 has nothing to do with the parameters entering the stack: -0.4 & 10 to control the PlotYRange definition.

As the program stops suddenly, the stack is expected to contain only the numerical values: -0.4 & 10 as argument causing PlotYRange to bug (same argument for all trials). Before that, the stack contained XMIN & XMAX as argument accepted of PlotYRange. And this is precisely XMIN which depends directly on X1. 

Following the program execution in the debug mode, we observe that this variables X1 is associated to the LocalVariable2. In the main program the correspondances for the 11 local variables are:
LocalVariable0 = Y1 
LocalVariable1 = Y2
LocalVariable2 = X1 
LocalVariable3 = X2
LocalVariable4 = Z1
LocalVariable5 = Z2
LocalVariable6 = P1
LocalVariable7 = P2
LocalVariable8 = XMIN
LocalVariable9 = XMAX
LocalVariable10 = XDELTA.

This is what I know for the moment. Any idea of what to investigate then ? How to proceed ?

Thanks for your advice.

LdBeth, 2025-04-01:
Since this is a very long program, do you know how to load it in the desktop simulator so I can start debug it? I cannot get mine working because the function library directory seems to be reset every time the sim starts up.

Wiljea, 2025-04-02:
@LdBeth 

Have you tried to copy and paste it to the screen of the simulator ? May be it will be easier if I reformat it as a single paragraph like the following (without comments):

```rpl
« Swap List→ Drop Duplicate2 if > then Swap end 2 →List Swap List→ Drop Duplicate2 if > then Swap end 2 →List Swap List→ Drop Rot List→ Drop 0 0 0 0 0 0 0 → Y1 Y2 X1 X2 Z1 Z2 P1 P2 XMIN XMAX XDELTA « X1 Y1 Max 'Z1' Store X2 Y2 Min 'Z2' Store X1 Y1 Min 'XMIN' Store X2 Y2 Max 'XMAX' Store XMAX XMIN - 'XDELTA' Store XMIN XDELTA 20 ÷ - 'XMIN' Store XMAX XDELTA 20 ÷ + 'XMAX' Store XMIN UnitValue XMAX UnitValue PlotXRange -0.4 10 PlotYRange "Concordance TEST Illustration" XMIN 0.08 XDELTA × + UnitValue 10 ℝ→ℂ DrawText X1 UnitValue 7.8 ℝ→ℂ X2 UnitValue 7.8 ℝ→ℂ DrawLine "|" X1 UnitValue 8.5 ℝ→ℂ DrawText "|" X2 UnitValue 8.5 ℝ→ℂ DrawText Y1 UnitValue 6.2 ℝ→ℂ Y2 UnitValue 6.2 ℝ→ℂ DrawLine "|" Y1 UnitValue 7 ℝ→ℂ DrawText "|" Y2 UnitValue 7 ℝ→ℂ DrawText if Z1 Z2 = then Z1 UnitValue 4.7 ℝ→ℂ Z2 UnitValue 4.7 ℝ→ℂ DrawLine "|" Z1 UnitValue 5.5 ℝ→ℂ DrawText "IY ∩ IX = {" Z1 + " " + Z2 + "} " + XMIN UnitValue 4 ℝ→ℂ DrawText "Concordance in one point only of" XMIN 0.01 XDELTA × + UnitValue 2.5 ℝ→ℂ DrawText "IX={" X1 + " " + X2 + "}; IY={" + Y1 + " " + Y2 + "}" + XMIN 0.01 XDELTA × + UnitValue 1 ℝ→ℂ DrawText end if Z1 Z2 < then Z1 UnitValue 4.7 ℝ→ℂ Z2 UnitValue 4.7 ℝ→ℂ DrawLine "|" Z1 UnitValue 5.5 ℝ→ℂ DrawText "|" Z2 UnitValue 5.5 ℝ→ℂ DrawText "IY ∩ IX = {" Z1 + " " + Z2 + "} =>" + XMIN 0.04 XDELTA × + UnitValue 4 ℝ→ℂ DrawText if X1 Z1 ≥ then Z2 Z1 X1 Max - X2 X1 - ÷ 100 × →Num 2 Negate Round 'P1' Store Z2 Y2 Min Z1 - Y2 Y1 - ÷ 100 × →Num 2 Negate Round 'P2' Store P1 "% of IX = {" + X1 + " " + X2 + "};" + XMIN 0.08 XDELTA × + UnitValue 2.5 ℝ→ℂ DrawText  P2 "% of IY = {" + Y1 + " " + Y2 + "}" + XMIN 0.08 XDELTA × + UnitValue 1 ℝ→ℂ DrawText end if Y1 Z1 ≥ then Z2 X2 Min Z1 - X2 X1 - ÷ 100 × →Num 2 Negate Round 'P1' Store Z2 Z1 Y1 Max - Y2 Y1 - ÷ 100 × →Num 2 Negate Round 'P2' Store P1 "% of IX = {" + X1 + " " + X2 + "};" + XMIN 0.08 XDELTA × + UnitValue 2.5 ℝ→ℂ DrawText P2 "% of IY = {" + Y1 + " " + Y2 + "}" + XMIN 0.08 XDELTA × + UnitValue 1 ℝ→ℂ DrawText end else if Z1 Z2 > then "IY ∩ IX is empty " XMIN 0.04 XDELTA × + UnitValue 5.5 ℝ→ℂ DrawText "=> No concordance for" XMIN 0.08 XDELTA × + UnitValue 4 ℝ→ℂ DrawText "IX = {" X1 + " " + X2 + "};" + XMIN 0.08 XDELTA × + UnitValue 2.5 ℝ→ℂ DrawText "IY = {" Y1 + " " + Y2 + "}" + XMIN 0.08 XDELTA × + UnitValue 1 ℝ→ℂ DrawText end end » »
```
I just verify the copy & paste action and it works fine for me (my simulator v0.9.2-14-3394 runs in the Fedora-Remix environment under Win10 ). Thanks for checking at least if you can reproduce the bug. Your help is very much appreciated.

c3d, 2025-04-05:
I won't go into the details of the debugging session, which was a bit intense and required me to add quite a bit of instrumentation, but the bottom line is that I believe this is a simple case of GC pointer mis-management in function `expression::simplify` because I did not acknowledge that `is_simplifiable` could cause a garbage collection. The reason it can is because constants contain units, and parsing units requires memory allocations, which can GC. So in some very rare cases, a GC would occur during any arithmetic operation involving constants and cause some bad side effects.

I need to re-validate everything with your test cases, though, so not 100% certain this is solved, but let's say 90% confident this is it.

c3d, 2025-04-06:
OK, your specific problem was not actually solved by the (real) bug above. So need to find what the next bug is.

c3d, 2025-04-06:
Reduced test case:

```rpl
20 30 XRange
```

Changing the test description.

commit 200eb3a9 (2025-04-06): graphics: Avoid testing `ppar` validity before end of parsing

When parsing `ppar`, we cannot run through `check_validity()` before
end of parsing. Otherwise we can fail the test `xmin<xmax` when
switching between two X ranges where the intermediate state fails that
check.

Fixes: #1439
closed 2025-04-06 by c3d