c3d/db48x #1440: Computation error with "ln" of large integer

jcbenoist, 2025-03-30:
https://fr.wikipedia.org/wiki/Approximation_de_%CF%80 gives some formula to approximate Pi. But Db48x does not compute the correct approximation, even with 100 digits precision.

For example : <math>\frac{\ln(640320^3+744)}{\sqrt{163}}</math> fails to gives correct result. 640320^3+744 is correct, but taking "ln" gives false decimals compared to DM42 (starting at 15th decimal). 

But computation is correct if "640320" is entered as real number, not integer.

This formula, and perhaps others, should be added to regression tests.

Context : DB48x 0.9.2, 100 digits precision, but same problem with 24 or 32 digits.

LdBeth, 2025-04-01:
Using 100 precision on DM42n I have `640320 3 ^ 744 + ln 163 sqrt /` get result 3.14159...83279 72661 that all the digits matches. Could be a hardware relevant issue.

jcbenoist, 2025-04-02:
@LdBeth I don't quite understand your answer. DM42n with DB50x ?  And you don't display the false digits. I don't understand why you talk about hardware , the emulator gives the same results.

The right computation is : 3,14159 26535 89793 23846 26433 83279+
DB50x, with 100 or 32 precisions, gives (* shows start of false digits)
 3,14159 26535 89793 *02276 10793 10900..

Computation with 640320.0000 (real) instead of integer gives correct results (not hardware IEEE floats)

c3d, 2025-04-02:
I cannot reproduce with either v0.9.2 or the current v0.9.3 candidate.
```
'ln(640320^3+744)/sqrt(163)' 32 PREC 32 SIG EVAL
```

gives me: `3.14159 26535 89793 23846 26433 83279 7`

[image]

I've tried a dozen variations, and could not find any combination where I would get the bad digits.

You mention the conversion between integer and decimal. There was a recent fix in that area, but it only impacts numbers that ended in 0 or 00 that could be incorrectly normalized when converted to decimal.

What I could reproduce is that the last digits in 24-digit mode are incorrect, which needs fixing.

c3d, 2025-04-02:
Oh, I see. Let me investigate if the `CUBED` command gives a bad result somehow.

LdBeth, 2025-04-02:
> But I cannot exhibit it with ToDecimal

The problem is also present with ToDecimal. `987654321 sq ->NUM` gives inaccurate result compared to `987654321 2 ^ ->NUM`.

c3d, 2025-04-02:
OK. The problem is [this line](https://github.com/c3d/db48x/blob/a5876494975bca2511896334540a2358277c82a6/src/decimal.h#L158): it should be `div >= 1000` and not `div > 1000`. Otherwise, we miss the loop for the last 3 digits.

c3d, 2025-04-02:
Obviously, this will be fixed in 0.9.3

jcbenoist, 2025-04-03:
Well done !

closed 2025-04-06 by c3d