c3d/db48x #1441: Bug converting vector from rectangular to polar

BadtoysWEF, 2025-03-30:
When not using complex number formats but vector between brackets [A B]  ->cylindric produces a non-sense angle. Sometimes happens sometimes it doesn't.

LdBeth, 2025-04-01:
`ToCylindrical` takes a 3D vector, and does not do rectangular to polar conversion. Can you give a concrete example?

c3d, 2025-04-02:
@BadtoysWEF I do not understand how you get the function to produce a non-sense angle. What I get is `Bad argument type`.

Could you give a short example of code. You can put on the stack a small RPL program that exhibits the problem, and then use the copy-paste feature of the simulator.

BadtoysWEF, 2025-04-02:
I haven't been able to reproduce the problem again.  I reset the calculator with a clip and now it looks like it is working. I have been having issues when I purge variables that I used with the solver. I must go Up directory and back again to execute or see what is inside the variables. The problem happened after that, but as I said, I can't reproduce the problem again. (I was using the triangle solver). Anyway, Thanks for your prompt answer. I'm longing for the 1.0.0 version.

closed 2025-04-02 by BadtoysWEF

LdBeth, 2025-04-02:
Sometimes calling GarbageCollector can solve weird behaviors, especially if you are using on the hardware