c3d/db48x #1445: Sqrt result precision went bad

LdBeth, 2025-04-06:
It is a regression starts in 0.9.3, sqrt(5) for example gives 5.00000 00000 00000 00000 00281 5048 in prec 30 when displayed in full, where 0.9.0 or 0.9.2 it would not introduce trailing digits at all.

LdBeth, 2025-04-06:
Also I can confirm that cbrt seems to be unaffected.

LdBeth, 2025-04-07:
By bisecting when the result of `36 sqrt` has been changed, this is likely caused by 68bd98b

LdBeth, 2025-04-07:
It appears to me that `s = std::min(s, e);` does not make much sense here

c3d, 2025-04-08:
Good catch @LdBeth.

The rationale for `s = std::min(s, e)`: At this point, `e` indicates how many _kigits_ (base-1000 digits) to consider for the comparison. `s` is the number of kigits in the inputs. We cannot compare beyond any of these, because the data beyond that is invalid.

However, you are correct that this does not make much sense because we want to go into the `i < xs` and `i < ys` conditions where we zero out the kigit anyway. Otherwise, the comparision ends up not seeing the last digits as zeroes.

Fix under test.

closed 2025-04-11 by LdBeth