c3d/db48x #1448: Bug in the function "StdRnd"

Wiljea, 2025-04-10:
The problem is encountered in **DB48X version v0.9.3** (stable branch). While I was recently developing an example of uncertainty calculation, I calculated the following result for the couple of values of m & Δm (as in M=m±Δm) for the mass of a glass marble using Range operations (this intermediate result is correct):
m ± Δm = 5.36248084521_g ± 0.11125683347_g
where I want to:  **i)** round the standard uncertainty (Δm) to 2 significant figures (SF) and then **ii)** use the internal function **StdRnd** to round the principal value m to show the same number of decimals as Δm. Then, I did:
**ex.1:** (failed using actual implementation)
```rpl
5.36248084521_g 0.11125683347_g -2 Round Duplicate Unrot StdRnd Swap
@ Expecting [ 5.36_g 0.11_g ]
```
And I got the following wrong answer for the rounding of m:
5.36248 g 0.11 g

If I'm looking back to an email exchange with @c3d (February 9th, 2015), my advice then was initially to define this **StdRnd** function as the following RPL code (note that, at the time, my implementation differed from @c3d's one, as stressed in the email):
```rpl
« → X UsX « X UsX UVAL XPON X UVAL XPON - 2 - ROUND » »
```
Which gives the expected result in the present case. Indeed, if you do:
**ex.1:** (succeeded using my implementation)
```rpl
5.36248084521_g 0.11125683347_g -2 Round Duplicate Unrot « → X UsX « X UsX UVAL XPON X UVAL XPON - 2 - ROUND » » Eval Swap
@ Expecting [ 5.36_g 0.11_g ]
```

This bug is surprising here because otherwise the actual implementation of **StdRnd** seems to work correctly (as mine) for other examples of standard uncertainty calculations like:
**ex.2:** (a basic one):
```rpl
3.141592654_m 0.000045_m StdRnd
@ Expecting [ 3.14159 3 m ]
```
And
**ex.3:** (a more advanced one: the correct rounding of the calculated constant µ0=1.25663 70612 697E-6 N/A↑2):
```rpl
14 SIG α=0.00729735256434 µ0='CONVERT(4·Ⓒπ·α·Ⓒℏ/(Ⓒqe^2·Ⓒc);1_N/A^2)' Eval Usµ0=2.0E-16 N/A↑2 µ0 Usµ0 StdRnd
@ Expecting [ 1.25663 70612 7E-6 N/A↑2 ]

I didn't check how **StdRnd** is implemented, but I suggest that the corrected version will need to satisfy all of these 3 examples. Because of this bug I began to doubt about the validity of the related command **RelRnd** that I suggested to be implemented as (« → X UrX « X DUP UrX * UVAL XPON X UVAL XPON - 2 - ROUND » ») but, so far, no problem was detected. 

Indeed, I checked **RelRnd** with the correct rounding of the calculated constant ε0=8.85418 78187 894E-12 F/m:
```rpl
14 SIG µ0=1.25663 70612 7E-6 N/A↑2 ε0='CONVERT(1/(Ⓒc^2·µ0);1_F/m)' Eval Urε0=1.6E-10 ε0 Urε0 RelRnd  
@ Expecting [ 8.85418 78188E-12 F/m ]
```
Which is OK.

c3d, 2025-04-12:
OK, this one is relatively subtle.

The problem is that `0.11_g` is not a base unit, the base unit is the `kg`, so this converts to `0.00011_kg`, and from there the exponents are off.

Relative to the original proposal, I obviously added code that converts the `UsX` to its `UBASE`. From memory it was to correctly deal with cases like `1_s/s` and converting them to `1`.

It looks like we really want to convert to the same unit as the first argument.

Wiljea, 2025-04-18:
Done in v0.9.3

commit 07e42639 (2025-04-13): functions: Adjust unit conversion in `StdRnd`

The `StdRnd` command was just looking at the values of its arguments,
but after a conversion to base unit for the second one that would turn
`0.11_g` into `0.00011_kg` (the kg being the `UBASE` unit). This would
give incorrect results.

Instead, perform a conversion from second argument to the type of the
first one. This also checks unit consistency in a more solid manner.

Also added some unit tests for these specific cases and for error
cases (e.g. passing a unit to second argument of `RelRnd`).

Also added the commands to the `PartsMenu` and re [... cut]
closed 2025-04-18 by Wiljea