c3d/db48x #1449: Suggestion: display global units for collective objects whose elements have same units

Wiljea, 2025-04-10:
This comes from a practical problem in the context of limited screen space. In physics, we often encounter quantities with quite complicated units like R=8.31446261815_J/(mol·K). If such quantities or derived values pertain to a: list, vector, matrix, range or complex number (in cartesian form), this consumes some prohibitive space on the simulator screen, either on the stack or in graphical representations. So I was wondering that if a flag "Collective Units" would be pertinent to allow formats like, for instance: [ 60.12 88.23 90.36 ]_J/(mol·K), etc. I agree that this is unusual, but frankly useful to avoid redundancy and replace [ 60.12_J/(mol·K) 88.23_J/(mol·K) 90.36_J/(mol·K) ].

c3d, 2025-04-10:
This seems a bit impractical, because it would require a comparison of all units in a vector, and presumably running `UBASE` on them for each display operation so that we identify `J/(mol*K)` and `J/(K*mol)`.

IMO this is prohibitively expensive in particular on physical devices..

Wiljea, 2025-04-10:
Closed as being unreasonable.

closed 2025-04-10 by Wiljea