c3d/db48x #1452: GraphicResultDisplay has no effect, but GraphicStackDisplay also contrtols the Result line.

be-philippe, 2025-04-14:
Versions 0.9.3 and 0.9.4 certainly, probably since version 0.8.8.

be-philippe, 2025-04-14:
I fixed it on my PC. Do you accept code contributions, and if yes, what are the rules ?

be-philippe, 2025-04-15:
In
https://github.com/c3d/db48x/blob/2f11a6e38f0cbcf65032f7208f7cbd80ed4eb4cb/src/stack.cc#L165-L167

- second `sgraph` in line 165 should be `rgraph`

- line 167 must come before line 165

c3d, 2025-04-24:
Of course, I accept code contributions. Please submit a PR.

As for commit messages, try to follow the following rules (but I'll reformat otherwise):

1. 70 characters wide max
2. First line is a general description, begins with a component, e.g. `stack: use GraphicResultDisplay for result`
3. Next lines give details of the problem and how you fixed it.
4. If appropriate, finish with a `Fixes` tag, e.g. `Fixes: #$1452` in this case
5. Please add a `Signed-off-by` line with a name and an email you are comfortable sharing
6. Commit this to a branch of yours, then submit a pull request.

I should write this in a `CONTRIBUTING.md` document.

c3d, 2025-04-24:
@be-philippe In that specific case, I wanted to also add some testing. Please review commit:2baf4d86858df6739cbe066aede9b36b7450f922

closed 2025-04-28 by c3d