c3d/db48x #1453: Object info show "zu bytes" instead of the object size, only on DMCP

be-philippe, 2025-04-14:
For version 0.9.4:

From dynamic stack, Shift F6 displays an information box about the current stack object.
On MacOs sim, the display is correct ("13 bytes decimal", for example).
On DM42, the display shows "zu bytes".
Not tested on DM32 nor DM42n, but probably the same problem.

Very probably related to the recent change to nanolibc, which doesn't support C99 size specifiers like z, hh or ll in printf and derived functions.

LdBeth, 2025-04-15:
Actually, the build using ARM GCC 14.2 either crashes into "Production Diagnostics" or "Exception" when `Info` command been used, ever since the 0.9.2 version I built, that I only just realize what does it supposed to do.

LdBeth, 2025-04-15:
The relevant code
https://github.com/c3d/db48x/blob/2f11a6e38f0cbcf65032f7208f7cbd80ed4eb4cb/src/user_interface.cc#L4594

#1431

be-philippe, 2025-04-15:
There are also several uses of %llu and %lld in the code. Most are in record() or test, thus not impacting the DM42.

There are 2 in polynomial.cc, impacting the display of polynomials (exponents become lu):
https://github.com/c3d/db48x/blob/2f11a6e38f0cbcf65032f7208f7cbd80ed4eb4cb/src/polynomial.cc#L1237
https://github.com/c3d/db48x/blob/2f11a6e38f0cbcf65032f7208f7cbd80ed4eb4cb/src/polynomial.cc#L1299

prismv, 2025-04-15:
My view is that it would be good to allow development of support of newer ARM GCC versions, although with low priority. The immediate benefit to use nanolib was a somewhat reduced program size.

The findings here show that also for the 'stable' ARM GCC 10.3.1 the change to nanolib requires some more fixes like 25cd85e6. Because @be-philippe mentioned unavailabilty of z (for this particular issue relevant) and hh formats,
and to be fixed ll problems in polynomial code, either the fix 25cd85e6 needs to be generalized or some other solutions like e.g. https://github.com/weiss/c99-snprintf have to considered.

If needed an ad-hoc fix would be substitution of 2e5e1812, at best (IMHO) with a change like discussed in https://github.com/c3d/db48x/issues/1430#issuecomment-2795224546, i.e. specific makefile variable definitions with `+=` instead of `=`.

prismv, 2025-04-15:
> The relevant code
> 
> [db48x/src/user_interface.cc](https://github.com/c3d/db48x/blob/2f11a6e38f0cbcf65032f7208f7cbd80ed4eb4cb/src/user_interface.cc#L4594)
> 
> Line 4594 in [2f11a6e](/c3d/db48x/commit/2f11a6e38f0cbcf65032f7208f7cbd80ed4eb4cb)
> 
>  else if (shift) 
> [#1431](https://github.com/c3d/db48x/pull/1431)

A simple fix for this might be:

```diff
diff --git a/src/user_interface.cc b/src/user_interface.cc
index b1d25e15..667c761e 100644
--- a/src/user_interface.cc
+++ b/src/user_interface.cc
@@ -4614,8 +4614,8 @@ bool user_interface::handle_editing(int key)
                     {
                         char sizebuf[40];
                         char valbuf[40];
-                        snprintf(sizebuf, sizeof(sizebuf), "%zu bytes %s",
-                                 size, obj->fancy());
+                        snprintf(sizebuf, sizeof(sizebuf), "%lu bytes %s",
+                                 (uint32_t)size, obj->fancy());
                         bin->render(valbuf, sizeof(valbuf));
                         draw_message("Object info", sizebuf, valbuf);
                         wait_for_key_press();
(END)
```

prismv, 2025-04-15:
> There are also several uses of %llu and %lld in the code. Most are in record() or test, thus not impacting the DM42.
> 
> There are 2 in polynomial.cc, impacting the display of polynomials (exponents become lu):
> 
> [db48x/src/polynomial.cc](https://github.com/c3d/db48x/blob/2f11a6e38f0cbcf65032f7208f7cbd80ed4eb4cb/src/polynomial.cc#L1237)
> 
> Line 1237 in [2f11a6e](/c3d/db48x/commit/2f11a6e38f0cbcf65032f7208f7cbd80ed4eb4cb)
> 
>  r.printf("%llu", exponent); 
> 
> [db48x/src/polynomial.cc](https://github.com/c3d/db48x/blob/2f11a6e38f0cbcf65032f7208f7cbd80ed4eb4cb/src/polynomial.cc#L1299)
> 
> Line 1299 in [2f11a6e](/c3d/db48x/commit/2f11a6e38f0cbcf65032f7208f7cbd80ed4eb4cb)
> 
>  snprintf(exptxt, sizeof(exptxt), "%llu", exponent);

The fix would be very similar to https://github.com/c3d/db48x/commit/25cd85e69275c3b8f6f5a9892ac931720d530157 in `decimal.cc`, but I suppose that this would require some more basic changes, when refactoring e.g. printf for rendering. So quite unsure how to add code here with my very moderate C++ knowledge.

LdBeth, 2025-04-15:
> A simple fix for this might be:
> 
> diff --git a/src/user_interface.cc b/src/user_interface.cc

At least the good thing is Info command no longer crashes after this been applied.

prismv, 2025-04-17:
See bd377d5021a374cc759979f5f4047e135054a5c8

prismv, 2025-04-17:
@be-philippe would be good to change the issue title to include bugs in polynomials.cc, because info bug is fixed.

c3d, 2025-04-23:
Fixed in commit:bd377d5021a374cc759979f5f4047e135054a5c8 which is in 0.9.5, so closing that issue.

prismv, 2025-04-23:
@c3d Do you want a new issue for "and exponent in polynomials shown as "lu", both only on DMCP" part also described in this issue?

c3d, 2025-04-24:
Fixed the polynomial case as well, but ran into a small bug where we were truncating exponents to 32 bits while building polynomials.

c3d, 2025-04-24:
@prismv No need for a new issue, since the problem is already fixed.

commit 368e8cca (2025-04-24): utils: Add routine to render u64 and i64

The nano version of libc does not support `%llu`. There were two
remaining `%llu ` formats in polynomials, as pointed out by user
@be-philippe.

Fixes: #1453
closed 2025-04-27 by c3d