c3d/db48x #1455: Crashing when I tried to subtract two vectors DB50x on DM42n

BadtoysWEF, 2025-04-16:
Hello I'm having this issue when [ 3 2 1.7º] [2 1 2.5º] -.

The calculator crash and shows the following message:

Oops, I've crashed! Please try to reproduce it, then report...

[image]

Thanks!!!!

PD. Would you port this project to the Hp Prime V2??? It would be great!

Thank you

LdBeth, 2025-04-16:
Also similar crash can happen with `Dot` when the two inputs are 3D and one contains degree unit.

LdBeth, 2025-04-16:
There are multiple potential bugs there, the first would be

https://github.com/c3d/db48x/blob/2f11a6e38f0cbcf65032f7208f7cbd80ed4eb4cb/src/array.cc#L231

I think it might not make much sense to classify a vector like `[2 1 2.5º]`, that gives `angles == 4`, as `Cylindrical` at first place, unless we can apply a different conversion function from the case `angles == 2`

c3d, 2025-04-23:
I have confirmed the issue, but unlike @LdBeth, I do not reproduce the problem in the simulator. Not sure why.

c3d, 2025-04-23:
OK, I agree with @LdBeth that it makes little sense to have two distinct formats for cylindrical coordinates. Removing the case `angles == 4` everywhere. There would be no good convension or where the `r` argument is in that case.

commit d046944c (2025-04-24): array: Fix issue in `+` and `-` on vectors with angles

When adding or subtracting vectors with angles in the last coordinate,
we were mistakenly identifying them as cylindrical vectors. The case
`angles == 4` was actually incorrect, i.e. vector `[ 1 2 3_° ]` is not
a valid cylindrical vector.

Fixes: #1455
closed 2025-04-27 by c3d