c3d/db48x #1456: Suggestion: Base units for English system

BadtoysWEF, 2025-04-16:
It would be nice to have a function like base but instead of metric units, convert to base English units like ft, lb, ...
Thanks!

c3d, 2025-04-24:
There is no such thing as "base English units". Would you pick `ft` or `yd`? Would you pick `oz` or `lb` or `ton`?

There are two solutions without modifying the firmware:

1. You can simply use `UFact` to factor out a specific unit. For example, `1_N
   1_lb UFACT` will give `¹⁰⁰ ⁰⁰⁰ ⁰⁰⁰/₄₅ ₃₅₉ ₂₃₇ lb·m/s↑2`.  If you factor with
   `1_lb*ft`, you get `¹²⁵ ⁰⁰⁰ ⁰⁰⁰ ⁰⁰⁰/₁₇ ₂₈₁ ₈₆₉ ₂₉₇ ft·lb/s↑2`.
2. You can redefine units in `config/units.csv` to change what is the base units for you.

For example, the current definition includes the `kg` as the base unit (it's defined using itself):

```
    "kg",       "1_kg",                 // Kilogram (SI base unit)
    "g",        "1/1000_kg",            // Gram
    "t",        "1000_kg",              // Metric ton
    "ct",       "200_mg",               // Carat
    "mol",      "1_mol",                // Mole (quantity of matter)

    // Imperial units
    "lb",       "45359237/100000_g",    // Avoirdupois pound
    "oz",       "1/16_lb",              // Ounce
    "dr",       "1/256_lb",             // Drachm
    "stone",    "14_lb",                // Stone
    "grain",    "1/7000_lb",            // Grain (sometimes "gr")

```

But we can instead define the `lb` as being defined from itself, and then we need to override the `kg` definition to define it from the `lb` instead of the other way around. Of course, we need to do the same thing with `m` vs. `yd`. We can add this in a separate section called `UKBase` in the condiguration file, so that your `config/units.csv` contains:

```
"UKBases",

"lb",   "1_lb"
"g",    "100000/45359237_lb"
"kg",   "1000_g"
"yd",   "1_yd"
"m",       "10000/9144_yd",
```

Once you have done that, `1_N UBASE` gives you `¹²⁵ ⁰⁰⁰ ⁰⁰⁰ ⁰⁰⁰/₅₁ ₈₄₅ ₆₀₇ ₈₉₁ lb·yd/s↑2` which I believe is what you want.

c3d, 2025-04-24:
Marking a `done`, because I think there is a good solution without changing the code.

BadtoysWEF, 2025-04-24:
Thank you!! That was exactly what I needed.

closed 2025-04-24 by BadtoysWEF