c3d/db48x #1458: The WalkMan demo doesn't work anymore on DM42.

be-philippe, 2025-04-17:
Shows

**VariablesMenuExecute error:
Malformed local program**

Some other demo programs have the same problem.

prismv, 2025-04-17:
The demo works for me, if compiled with GCC 10.3.1. If you use newer versions like 14.x, you quite certainly face issue #1286.

Wiljea, 2025-04-20:
It works for me with DB48X 0.9.4-24-8a3c.

be-philippe, 2025-04-21:
Yes, I was using GCC 14.2.1 (latest release).

I did some quite extensive analysis, and the problem arises here:
https://github.com/c3d/db48x/blob/2f11a6e38f0cbcf65032f7208f7cbd80ed4eb4cb/src/locals.cc#L237-L242

Recent compilers keep the `p` variable in a register, and the increment to `p` done by the call to `leb128<size_t>(+p)` is lost.

With your example from issue #1286, the `nlen` from the first execution of the loop moves `p` one byte forward, to the name of the first local `'x'`, which the second execution of the loop returns as it's ASCII code, `120`. `p` points then to some irrelevant byte, thus the error message.

I've analyzed this code fragment extensively, and I find no fault in it, **so it really looks like a compiler bug**.

Introducing a working copy of `p` bypasses the problem (no more error message):
```
    // Skip names to get to program
    for (uint n = 0; n < names; n++)
    {
        object_p pp = +p;
        size_t nlen = leb128size(pp);
        nlen += leb128<size_t>(pp);
        p += nlen;
    }
```
but then the expression does nothing (with GCC 14.2.1).

I managed to install an older version of GCC on my Mac (12.2.1, the oldest one existing for Apple Silicon).

The original code encounters the same problem, but the patch seems to work (I've been able to execute most of the demo programs).

Alas, this increases the .pgm size from 644 212 bytes to 648 076 bytes.

be-philippe, 2025-04-21:
That "compiler bug" thing was still bugging me (how could such a bug remain undetected for several years ?), so I gave it more tought, and I found the actual bug (more an Undefined Behaviour) in these 2 lines of code.

From the compiler point of vue, `+p` is computed from `p`, but has no other link with it. As the `+p` object is transitory, and thus never used later, the compiler doesn't store it in ram, and the modification made by `leb128<size_t>(p)` is lost.

The actual correct code is:
```
    for (uint n = 0; n < names; n++)
    {
        object_p pp = +p;
        size_t nlen = leb128<size_t>(pp);
        p = pp + nlen;
    }
```
The compiler actually stores `pp` in ram before calling `leb128<size_t>(pp)`, and reads it back after.

This also fixes the bug with GCC 12.2.1, but GCC 14.2.1 still encounters  bugs elsewhere in the code.

prismv, 2025-04-21:
Thank you @be-philippe for your explanations and findings.

I understand that the argument with unary plus `+p` is needed to indicate that `inline Data *leb128(Data *&p)` has to be called, and similar calls like `leb128<size_t>(+names)` are used later on.

I have no real clue, if `*&p` arguments may have some substitutes in newer C++ standards, so that might 
explain some digression specifically with the arm cross-compiler eventually also related to optimizations in leb128.h for DM42?.

be-philippe, 2025-04-21:
I didn't check in details, but I guess the main difference with the simulator is that the latter one is compiled in debug mode, without optimisations.

be-philippe, 2025-04-21:
The only real solution is to fix all these occurences one by one.

Nevertheless, in the interim, one hack is to INLINE the calls to `leb128<size_t>(+...)`. As is, it would be quite brutal for the .pgm size, but it is possible to change the signature of the actual `leb128<size_t>(+...)` to avoid the `&*p` argument, and provide an `INLINE` thunk with the current signature to avoid the need to fix everything by hand.

The function with the modified signature:
```
template <typename Int = uint, typename Data>
NOINLINE Data *leb128(Int &res, Data *p)
// ----------------------------------------------------------------------------
//   Return the leb128 value at pointer
// ----------------------------------------------------------------------------
{
    const bool is_signed = Int(~0ULL) < Int(0);
    byte      *bp        = (byte *) p;
    Int        result    = 0;
    unsigned   shift     = 0;
    do
    {
        result |= Int(*bp & 0x7F) << shift;
        shift += 7;
    } while (*bp++ & 0x80);

    if (is_signed && (bp[-1] & 0x40))
        result |= Int(~0ULL << (shift - 1));
    res = result;
    return (Data *) bp;
}
```
and the thunk:
```
template <typename Int = uint, typename Data>
inline INLINE Int leb128(Data *&p)
// ----------------------------------------------------------------------------
//   Return the leb128 value at pointer
// ----------------------------------------------------------------------------
{
    Int result;
    p = leb128(result, p);
    return result;
}
```
With that, the compilation with GCC 14.2.1 works perfectly, for everything I have tested.

prismv, 2025-04-21:
Great, @be-philippe !

Had to define `#define NOINLINE  __attribute__((noinline))` in types.h, but then your proposal works for gcc 14.2, both for PC simulator and DM42 hardware.

c3d, 2025-04-23:
Thansk @be-philippe for the detailed analysis and validation of teh fix. I think it's worth applying these patches for GCC 14 support. Do you want to submit a PR to get the credit, or do you want me to do it?

That being said, I want to discuss your statement that this is undefined behaviour:

> That "compiler bug" thing was still bugging me (how could such a bug remain undetected for several years ?), so I gave it more tought, and I found the actual bug (more an Undefined Behaviour) in these 2 lines of code.

I see nothing in the standard that would make it undefined behaviour.

The + operator on GC pointers explicitly returns a reference to a non-const pointer to a const `Obj`, and that's a reference to the internal pointer:

```c++
        const Obj *&operator+()         { return (const Obj * &) safe; }
```

This overload is the one that must be picked-up for non-const pointers. The `leb128` function is also taking a reference to a non-const pointer as its `p` argument:

```c++
template <typename Int = uint, typename Data>
inline Int leb128(Data *&p)
```

The `p` variable is declared as `object_g`, which is a non-const GC pointer. Therefore, it is _illegal_ for the compiler to pick up the const variant of the overload…

> 
> From the compiler point of vue, `+p` is computed from `p`, but has no other link with it.

… therefore that statement seems false to me. It has a well-defined link to it: it is a mutable reference to the internal pointer. This means that the compiler should either have a pointer to the pointer in the register, or elide pointers only if it then updates the target pointer (with sufficient inlining, it is permissible to elide pointers for a reference).

> As the `+p` object is transitory,

… a transitory reference, not a transitory pointer

> and thus never used later, the compiler doesn't store it in ram, and the modification made by `leb128<size_t>(p)` is lost.

I believe this is clearly a bug. However, it may be a result of the removal of two-stage overload resolution in GCC13, see https://gcc.gnu.org/gcc-13/porting_to.html. I have not declared the move operators, so maybe an implicit move constructor gets in the way. I need to dig further.

> The actual correct code is:
>
> ```
>     for (uint n = 0; n < names; n++)
>     {
>         object_p pp = +p;
>         size_t nlen = leb128<size_t>(pp);
>         p = pp + nlen;
>     }
> ```
>
> The compiler [...]

c3d, 2025-04-23:
Changed title from "_The WalkMan demo doesn't work anymore on DM42_" to more closely match what we now understand of the problem

c3d, 2025-04-23:
Marking it as `duplicate` of #1286

be-philippe, 2025-04-24:
I'll submit a PR.
Do you want one for just this, or should I also include some other small fixes ?

be-philippe, 2025-04-24:
The problem can't come from the change in GCC 13, it's already there in GCC 12.2.1.

be-philippe, 2025-04-24:
Recompiling the DM42 version with debugging information without any optimization, and objdumping the .elf file shows that the compiler correctly invokes the right operator+().

That's said, this syntax is just a fancy way to call a member function, which here happens to return a reference to a pointer to some byte.

We as humans know that said pointer is the significant part of the gcp object, but how could the compiler know that ?

I'm not even sure why my INLINE thunk works, nor if it'll still work in GCC 15, ...

prismv, 2025-04-24:
How ever this issue will be solved: Thank you both for the explanations regarding the operator overloading. Definitely my C++ knowledge level is improving by reading your discussion only.

c3d, 2025-04-24:
> Recompiling the DM42 version with debugging information without any optimization, and objdumping the .elf file shows that the compiler correctly invokes the right operator+().
>
> That's said, this [...]

It doesn't know. What it "knows" is that we return a _reference_, and that we then pass that reference as an argument to the `leb128` function. So the pointer update in `leb128` should update the original pointer through the reference. It's a bit as if we were passing a pointer. If the compiler chooses to do something else, it seems wrong to me.

> 
> I'm not even sure why my INLINE thunk works, nor if it'll still work in GCC 15, ...

be-philippe, 2025-04-25:
Is this a better explanation ?

The compiler considers functions one at a time.

When compiling `EVAL_BODY(locals)`, `+p` is just a call to the `gcp.operator+()` function returning a reference to some pointer.
That reference is provided to `leb128<size_t>(const object *&)`, which updates the pointer in RAM.

Analyzing each function separately, the aliasing between `p.safe` and `+p` is invisible to the compiler.

Consequently, for `EVAL_BODY(locals)`, the optimizer is allowed to keep both in separate registers. It saves `p.safe` back to RAM after each modification, needed because of the call to a non-`const` function on `p`.

Without the optimization initailizing `+p` once in a registry and keeping it there afterwards, all is fine.

But the language doesn't specify how to handle that case, so Undefined Behaviour, which leads to bugs appearing with a compiler update.

commit 69746581 (2025-05-06): general: bug with recent compiler (GCC >= 12.2.1)

With code optimized by a recent GCC compiler, the value of some `gcp`
pointer is kept in a CPU register, and the modification to that
same pointer through it's `operator+()` function provided to
`leb128<size_t>()` may be lost.

Explicitely returning the modified pointer from a `leb128<size_t>()`
with a different signature fixes the base bug.

A temporary hack is provided with a forced INLINE thunk with the
previous signature.

Fixes: #1458, #1286
commit 9a9811e4 (2025-05-06): general: bug with recent compiler (GCC >= 12.2.1)

With code optimized by a recent GCC compiler, the value of some `gcp`
pointer is kept in a CPU register, and the modification to that
same pointer through it's `operator+()` function provided to
`leb128<size_t>()` may be lost.

Explicitely returning the modified pointer from a `leb128<size_t>()`
with a different signature fixes the base bug.

A temporary hack is provided with a forced INLINE thunk with the
previous signature.

Fixes: #1458, #1286

Reviewed-by: Christophe de Dinechin <christophe@dinechin.org>
closed 2025-05-26 by c3d