c3d/db48x #1460: Power consumption measurement shows no proper idle mode

prismv, 2025-04-19:
If one takes the measurement result shown in https://www.hpmuseum.org/forum/thread-20157-post-202635.html#pid202635 for granted, then db48x has no proper idle mode as can be seen e.g. in C47 that uses also DMCP `sys_timer_start`.

I hope the hint given in https://www.hpmuseum.org/forum/thread-20157-post-202670.html#pid202670 to relevant changes in other DM42 projects is of help.

be-philippe, 2025-04-21:
For the DM42 to go to low power, there are 2 rules:
1. ALL timers must be disabled;
2. In a call to sys_sleep().

prismv, 2025-04-22:
Seems to be solved since release of 0.9.5 - see e.g. some forum thread response from mateo and in particular a [message by @be-philippe](https://www.hpmuseum.org/forum/thread-20157-post-202813.html#pid202813) actually showing some measurement results.

closed 2025-04-22 by prismv