c3d/db48x #1465: Bug in derivative of ASIN involving units ('∂' seems broken in v0.9.5.1)

Wiljea, 2025-04-24:
With DB48X v0.9.5.1 it seems that the function '∂()' is now broken, while in the previous version v0.9.5 it worked perfectly. 

Inside a program calculating the quadratic superposition of uncertainty (see at the end), I was trying to derive the function: '0.5·Asin(y·Ⓒg/x↑2)' evaluated at y=10_m & x=19.31439 75209_m/s. Looking at the derivative:
```rpl
yv=10_m xv=19.3143975209_m/s '∂y(0.5·Asin(y·→Num(Ⓒg)/x↑2))|y=yv|x=xv' 
@ => "sin⁻¹ error: Argument outside domain"
```
A somewhat different error (probably units related) occurred also in the following:
```rpl
zv=1_m '∂z(Asin(z·(1_m)·(1_(1/m²))))|z=zv' eval eval @ => "∂ error: Bad argument type"
```

The fault was originally seen in the complete quadratic program, which is:
```rpl
«
	«
		iferr
			'ρ12' Recall
		then
			0 Duplicate Rot Store
		end
	»
	→ f yv σy xv σx ρ
	«
		'∂x(→Num f)|x=xv|y=yv'
		Evaluate Evaluate '∂y(→Num
		f)|y=yv|x=xv' Evaluate
		Evaluate
		→ dfdx dfdy
		«
			'→Num f|y=yv|x=xv'
			Evaluate Evaluate '(dfdx·σx)↑2+(dfdy·σy)↑2'
			Evaluate ρ
			if
				0 =
			then
				0
			else
				'ρ12' Recall σy
				× σx × 2 × dfdx
				× dfdy ×
			end + √
		»
	»
» ssFXY STO
```
Therefore what happened ? This program was working fine recently in DB48X v0.9.5:
```rpl
ρ12=0  'y·4/3·Ⓒπ·(x/2)³' 2.5_g/cm↑3 0.005_g/cm↑3 1.600_cm 0.005_cm ssFXY 
        @Expecting [ 5.36165 14621 3_g  0.05139 65752 97_g ]
```
But not anymore ! I rebuilt the simulator again & again without success ! I'm wondering if the state of my simulator got corrupted ...

c3d, 2025-04-24:
There was absolutely nothing between 0.9.5 and 0.9.5.1 that could possibly impact derivatives. The only changes impacted measurement of time for the `Wait` command and the user interface.

Wiljea, 2025-04-25:
Indeed, these issues were not version dependent. I have precisely identified and re-described the single remaining issue.

Wiljea, 2025-04-26:
I verify and the units management bug is still there.

Wiljea, 2025-05-01:
There are two cases that pose problems.
Following mail exchanges with @c3d, we incorporate the correct way to compute the derivatives, for instance: f 'x' ∂ 'x=xv' | 'y=yv' | Evaluate →Num and use v.9.6.
1) The program ssFXY (computing general quadratic superposition of uncertainties) for binary functions becomes the following with case which works without units and a failing case with units:
```rpl
« « iferr 'ρ12' Recall then 0 Duplicate Rot Store end » 0 0 0 → f yv σy xv σx ρ funits σxunits σyunits « f 'x' ∂ 'x=xv' | 'y=yv' | Evaluate →Num f 'y' ∂ 'y=yv' | 'x=xv' | Evaluate →Num → dfdx dfdy « 'f|y=yv|x=xv' Evaluate →Num Duplicate Duplicate UnitValue ÷ 'funits' Store σx Dup Uval / 'σxunits' STO IF σxunits 1_° == THEN σx →Radians Uval 'σx' STO END σy Dup Uval / 'σyunits' STO IF σyunits 1_° == THEN σy →Radians Uval 'σy' STO END '(dfdx·σx)↑2+(dfdy·σy)↑2' Evaluate ρ if 0 = then 0 else 'ρ12' Recall σy × σx × 2 × dfdx Evaluate × dfdy Evaluate × end + √ case funits 1 ° == then 1 r × →Degrees end funits 1 r == then 1 r × end end » » » 'ssFXY' STO

ρ12=0 'y↑x' 2. 0.01  3. 0.03 ssFXY 
      @Expecting [ 8. 0.20511 97055 42]
ρ12=0 'y↑x' 2._m 0.01_m  3. 0.03 ssFXY 
      @Failing [ 8._m 0.20511 97055 42_m] => "ln error : Inconsistent units"
```
2) The program ssf for unary functions becomes the following with 2 cases which works followed by 2 similar cases which don't:
```rpl
« 0 0 → f xv σx funits σxunits « 'f|x=xv' Evaluate Eval →Num Dup Dup Uval / 'funits' STO f 'x' ∂ 'x=xv' | Evaluate Eval →Num abs σx Dup Uval / 'σxunits' STO IF σxunits 1_° == THEN σx →Radians Uval 'σx' STO END σx × CASE funits 1_° == THEN 1_r * →Degrees END funits 1_r == THEN 1_r * END END » » 'ssf' STO

'exp(x/(1_m/s))' 2_m/s 0.05_m/s ssf 
      @Expecting [ 7.38905609893 0.369452804947 ] 
R=5_Ω C=2_F 'exp(-x/(R·C))' 2_s 0.05_s ssf 
      @Expecting [ 0.818730753078 0.004093653765 s/(Ω·F) ] Ubase => 0.004093653765
'exp(x·(1_s)/(1_m))' 2_m/s 0.05_m/s ssf 
      @Failing [ 7.38905609893 0.369452804947 ] 
      => 'x' "÷ error : Inconsistent units"
R=5_Ω C=2_F 'exp(-x·(1/R)/C)' 2_s 0.05_s ssf 
      @Failing [  0.818730753078 0.004093653765 s/(Ω·F) ] 
      => 'x' "÷ error : Inconsistent units"
```
It is strange that there is no difference in the last comparison between working and failing cases.

Wiljea, 2025-05-10:
In my last comment the "Inconsistent units" error of case 1. (for 'y↑x'). is understandable due to units incoherence of the output data that would be formed as a sum of 2 quantities: one having units m^2.97 & the second one with units m^3.03 which are evidently incompatible. Contrary to this impossibility, the correct approach is to view the second input number (Δx of x±Δx) as a small uncertainty affecting only the central value (x) of the exponent, as it is used to calculate the output value and not to determine its units. Therefore, the reasonable solution is to calculate the units of the output using only the central value (x): units of 'y↑x' are units of y to the power x. This is done in the following version of the program ssFXY:
```rpl
« « iferr 'ρ12' Recall then 0 Duplicate Rot Store end » 0 0 0 0 → f yv σy xv σx ρ funits σxunits σyunits Punits « IF f 'y↑x' == THEN σx Dup Uval / 'σxunits' STO σy Dup Uval / 'σyunits' STO IF σxunits 1 ≠ THEN Y X "The exponent X must not have units" DoError END xv Uval 'xv' STO yv Uval 'yv' STO σx Uval 'σx' STO σy Uval 'σy' STO σyunits xv ↑ 'Punits' STO END f 'x' ∂ 'x=xv' | 'y=yv' | Evaluate →Num f 'y' ∂ 'y=yv' | 'x=xv' | Evaluate →Num → dfdx dfdy « 'f|y=yv|x=xv' Evaluate →Num Duplicate Duplicate UnitValue ÷ 'funits' Store IF σxunits 1_° == THEN σx →Radians Uval 'σx' STO END IF σyunits 1_° == THEN σy →Radians Uval 'σy' STO END '(dfdx·σx)↑2+(dfdy·σy)↑2' Evaluate ρ if 0 = then 0 else 'ρ12' Recall σy × σx × 2 × dfdx Evaluate × dfdy Evaluate × end + √ case funits 1 ° == then 1 r × →Degrees end funits 1 r == then 1 r × end end IF f 'y↑x' == THEN Punits * Swap Punits * Swap END Rot Drop » » » 'SsFXY' STO

ρ12=0 'y↑x' 2._m 0.01_m  3. 0.03 ssFXY
    @Expecting [ 8. m↑3.   0.20511 97055 42 m↑3. ]
```
which is the correct answer. So, case 1. is ruled out: no more problem.

But the problem remains entirely for case 2. which is of a different nature and where it appears that the derivative commands doesn't manage units properly ! Indeed, in some calls the units simplification is correctly done but not in other cases: see the end of previous remark where instead of the required answer we obtain **the following error message**: 
**"÷ error : Inconsistent units".**