c3d/db48x #1466: Convertion between miles or mph to kph not working

BadtoysWEF, 2025-04-24:
Hi. I've discovered that converting from mph to ft/s or vice versa doesn't work, nor does converting from km/h to mph. Also, from mi to miUS shows inconsistent Units. It seems to me like a problem with mi definition.

Thank you!

c3d, 2025-04-26:
Hi @BadtoysWEF. It works perfectly for me.

Given that we recently had a discussion in #1456 about using UK units as base units, and given that it works perfectly for me, including with a `config/units.csv` that contains the modifications I suggested in #1456, I strongly suspect something wrong in your edits of that file.

Please attach your `config/units.csv` for inspection. Also, please describe an exact, step-by-step reproducer for you from either an empty state file or the original `Demo.48s` state file.

Thanks.

BadtoysWEF, 2025-04-26:
[units.csv](https://github.com/user-attachments/files/19925598/units.csv)

Hello!
Here is my Units file. Thank you so much for your help

LdBeth, 2025-04-27:
```
"Quick"

"in"    "="
"ft",	"="
```
missed a comma there

BadtoysWEF, 2025-04-27:
Thank you. I just corrected that, but the issue is still there.

BadtoysWEF, 2025-04-28:
I've deleted from my units file fps and miles and now it is working, however I don't understand what my mistake is.

c3d, 2025-04-28:
Two additional issues, one of which is a DB48x bug that I need to think about.

1. Your `CF` unit uses a DB48x keyword (ClearFlag). This causes a cascading failure while parsing the unit file. This is a bug. For now I renamed it to `CFt`, and then the `Civil` menu looks OK.
2. You defined the `hr` whereas the standard unit is `h`. There is no connection between `hr` and `h` in your unit file or in the internal definitions, so you cannot convert from `mi/hr` to `km/h`.

c3d, 2025-04-28:
If you define `hr` as `1_h` instead of `1_hr`, then it works.

c3d, 2025-04-28:
Here is the version of [units.csv](https://github.com/user-attachments/files/19948774/units.csv) that works for me. Closing the issue.

closed 2025-04-28 by c3d

BadtoysWEF, 2025-04-28:
Thank you!!! a couple of commas were making the problem. Now it is working like a charm!!