c3d/db48x #1471: Derivative can't receive its argument through a local variable

Wiljea, 2025-04-28:
This used to work fine in the previous version (0.9.5 stable) but doesn't anymore in the newest version v0.9.6 (stable):
```rpl
'x³' 2 « → f xv « '∂x(f)|x=xv' Evaluate Evaluate » »
  => 0 @Failing [ 12 ]
```
which obviously the wrong answer !

c3d, 2025-04-28:
There was a change ensuring that we don't pre-evaluate the argument of `∂x` prior to running the necessary formal transforms. This was done to make sure that we would get an `Unknown derivative` error (as expected) for `∂x(IP(x))`. This also means that `f` is not evaluated in the above example. Instead, a formal derivative of `f` is computed _before_ we do the `x=xv` replacement. The formal partial derivative of `f` relative to `x` is `0`, hence the result.

Now, what you want to do makes sense, but this is _not_ what the HP50G does. The HP50G also returns `0` for your code.

Fortunately, there is a relatively easy way to do what you want: while the order of evaluation in algebraic form is not what you want, you can still evaluate the arguments the way you want by using the stack-based approach:

```rpl
'x³' 2 « → f xv « f 'x' ∂ 'x=xv' | Evaluate » »
@ Expecting 12
```

To be pedantic, if you want to write it in a way that is also compatible with the HP50G, since the non-algebraic form of `|` will not accept `'x=xv'` as input, you would need to write it as:

```rpl
'x³' 2 « → f xv « f 'x' ∂ { x } xv + | Evaluate » »
```

However, then you run into a bug where the list input for `|` is not interpreted by DB48x in an HP50G-compatible way, a bug that will be fixed in the next release. So for now, you need the first form.

Again, the fact that this now returns 0 is not a bug, it's intentional.  I cannot think of a way to control the evaluation order in pure algebraic form. Usually, you control evaluation order of algebraic expressions using parentheses, but that's one case where parentheses won't apply. However, since you can precisely control evaluation order by not using the algebraic form, and since the current evaluation order matches what HP was doing, I think that's the best solution..

Wiljea, 2025-04-28:
Thanks ! I think I'll use the algebraic form ('x³' 2 « → f xv « f 'x' ∂ 'x=xv' | Evaluate » ») due to its more suggestive and readable form. We have to take advantage from the fact that the RPL of DB48X is somehow a more advanced evolution.

c3d, 2025-04-28:
Closing, since the new behaviour is similar to the HP50G.

closed 2025-04-28 by c3d