c3d/db48x #1473: SHOW uses the GrayPattern selected by the last program

be-philippe, 2025-04-29:
After interrupting the `Patterns` example program around the half of the screen, the digits displayed by the `Show` command are greyed.

c3d, 2025-05-06:
Now, I wonder if this is a bug, in the sense that I want the pattern to be configurable, and `Foreground` is as good a choice as any.

Either that, or we add a `ShowForeground` and `ShowBackground` setting. What do you think is the best?

be-philippe, 2025-05-07:
I find the current situation surprising, as I don't expect `SHOW` to depend on `Foreground` nor `Background` (it took me some time to guess the problem came from interupting the `Patterns` example). So I certainly wouldn't keep the current situation.

`ShowForeground` and `ShowBackground` settings would be adequate, although I don't see what use case there is.