c3d/db48x #1476: Loading db48x_qspi on DM42 "program is not targeted for this unit"

dylanmc, 2025-04-30:
This is my first time installing your system. I followed the instructions on your YouTube video about installing.  The `.pgm` loaded fine, but when it came time to the step "Load QSPI from FAT" I got a warning titled "Incompatible Program" - "Currently loaded program is not targeted for this unit. It may not work properly. We recommend to ... load another program".

The installation YouTube video doesn't show this, so I suspect either a build error or something wrong on my end. I decided not to chose the "Start DB48X Anyway (not safe)" menu option, afraid to brick my DM42.

LdBeth, 2025-04-30:
What is your firmware version? The program requires latest DMCP 3.25 from https://technical.swissmicros.com/dm42/firmware/DMCP/

be-philippe, 2025-05-01:
Did you also copy the `keymap.bin` to the DM42, and did you got the message ?
```
New KEYMAP installed
DB48X

Press any key to continue
```

be-philippe, 2025-05-01:
I just checked that reinstalling the original DM42 `keymap.bin` gives the "Incompatible Program" message.

c3d, 2025-05-06:
@be-philippe's suggestion is correct.

The recommended installation procedure now is to `tar xvfz` the release file at the top level of the mount point for the DM42's filesystem. That will also install configuration files, the help files, and so on.

c3d, 2025-05-06:
@dylanmc Could you be specific about the YouTube video (I believe that I describe the installation procedure several times). Just so that I can update the description to clarify.

dylanmc, 2025-05-06:
It's the installation video on this page: https://48calc.org/download

dylanmc, 2025-05-13:
Untarring the `.tgz` in the root directory, then loading the `qspi` (which loads the `.pgm`) worked fine.