c3d/db48x #1484: Inconsistent units in a derivative that should be coherent

Wiljea, 2025-05-16:
This error happens in the following:
```rpl
 xv=2_m '4/3·Ⓒπ·x³' 'x' ∂ 'x=xv' | Eval →Num 
@ => "| error: Inconsistent units"
```
It appears also in the equivalent following program:
```rpl
'4/3·Ⓒπ·x³' 2_m « → f xv « f 'x' ∂ 'x=xv' | Eval →Num » »
```
And also in the following program (at the derivative evaluation "|" in debug mode):
```rpl
'4/3·Ⓒπ·x³' 2_m 0.05_m 
« 0 0 0 → f xv σx funits σxunits fv « 'f|x=xv' Eval →Num 'fv' STO fv IF fv Typename "unit" == THEN 1 fv →Unit ELSE 1 END 'funits' STO f 'x' ∂ 'x=xv' | Eval →Num abs IF σx Typename "unit" == THEN 1 σx →Unit ELSE 1 END 'σxunits' STO IF σxunits 1_° == THEN σx →Radians Uval 'σx' STO END σx × CASE funits 1_° == THEN 1_r * →Degrees END funits 1_r == THEN 1_r * END END » »
@ => "| error: Inconsistent units"
```
It seems that in this program, when evaluating the derivative at xv=2_m, a sum of '4·π·4 m↑2' with '0_m↑3' leads to units incompatibility (suggestion : a zero unit object should be neutral in the case of additive operation, and should benefit from Uval to remain neutral).

evgaster, 2025-05-17:
The derivative is incorrect. 
just see what '4/3·Ⓒπ·x³' 'x' ∂ gives. It adds '+0·x³' to the derivative.
You then have x with different powers in the different terms. That makes the units inconsistent.
This is already the case before the substitution.

A zero unit object is not neutral. Its units carry meaning.
Would you consider a 1 with units neutral in a multiplication operation?

A temporary work around (until derivation is fixed) for this particular example would be an Eval before the substitution.

I very briefly looked at the "following program". Please, next time, put some effort in formatting and comments.

evgaster, 2025-05-17:
I haven't looked at the code for derivative (could not find it, yet). So I am speculating.
Probably it stumbles over the Ⓒπ and treads it differently from 4/3.
I suspect that is what leads to the '+0·x³'.

if u = 4/3, v = Ⓒπ and w = x³ then (u * v * w)' = u' * v * w + u * v' * w + u * v * w'.
We get the last term first. I suspect that it does drop the first. The second does get evaluated to 0·x³ but isn't dropped.
Changing the order (although addition is commutative) is a bit annoying.

Wiljea, 2025-05-17:
The derivative alone gives:
'4/3·Ⓒπ·x³' 'x' ∂ => '4·π·x²+0·x³'
and strictly speaking it's correct if you consider it as a product f(x)·g(x) where f(x)=4/3·Ⓒπ is a constant function & g(x)=x³, and where a simplification should have been made to send to 0 any term of 0 multiplied by anything.

evgaster, 2025-05-18:
I am very new to this db48 stuff, coming from a HP28S and HP50g.
A, perhaps stupid, question: which key sequence activates the substitution operator | in RPN mode?
I have been typing 'SUBSTITUTE' in Alpha mode.

c3d, 2025-05-21:
> I haven't looked at the code for derivative (could not find it, yet).

The core of that code would be [here](https://github.com/c3d/db48x/blob/02409d2f698e28fd62b3d9f49b06c8fb1f2fd04e/src/expression.cc#L3906)

> So I am speculating.  Probably it stumbles over the Ⓒπ and treads it
> differently from 4/3.  I suspect that is what leads to the '+0·x³'.

Indeed, units are treated differently during [simplification of multiplication operations](https://github.com/c3d/db48x/blob/02409d2f698e28fd62b3d9f49b06c8fb1f2fd04e/src/arithmetic.cc#L486), through the auto-simplify flag is [temporarily switched off](https://github.com/c3d/db48x/blob/02409d2f698e28fd62b3d9f49b06c8fb1f2fd04e/src/expression.h#L205) during rewrites for performance but also to avoid undesired rewrites. The rewrite engine is used during differentiation above.

>
> if u = 4/3, v = Ⓒπ and w = x³ then (u * v * w)' = u' * v * w + u * v' * w + u *
> v * w'.  We get the last term first. I suspect that it does drop the first. The
> second does get evaluated to 0·x³ but isn't dropped.  Changing the order
> (although addition is commutative) is a bit annoying.
>

It's not really a matter of order but a matter of not simplifying the multiplication by zero.

c3d, 2025-05-21:
> I am very new to this db48 stuff, coming from a HP28S and HP50g. A, perhaps stupid, question: which key sequence activates the substitution operator | in RPN mode? I have been typing 'SUBSTITUTE' in Alpha mode.

Most general is Shift-Shift-7 (SYM menu), then F6 (ALGBRA), then F6 gives you `|`

With 'X' on the stack (or any expression), the tool key brings you the SYM menu. So it would be something like A then F6, and then you get `|` with F6.

Or in Alpha mode, Shift-Shift-6

Or `WHERE`, which is shorter than `SUBSTITUTE`

c3d, 2025-05-22:
It's really weird, because variations of

``` rpl
'4/3·Ⓒπ·x³' 'x' ∂
```

work better:

``` rpl
'4/3·x³' 'x' ∂      @ Works: 4*x^2
'4·Ⓒπ·x³' 'x' ∂     @ Works: 12*pi*x^2
```

In non-symbolic mode, it produces `'12.56637 06144·x²+0.·x³'`, which is really weird. The `0.` does not match what I would expect.

c3d, 2025-05-22:
Trying variations, I noticed that `'A·Ⓒπ·x³' 'x' ∂` correctly eliminates the `0` and produces `'A·Ⓒπ·x³' 'x' ∂`, but that `'(A÷B)·Ⓒπ·x³' 'x' ∂` also does not eliminate the zero and produces `'3·A÷B·π·x²+0÷B²·π·x³'`.

A true division (`'(4÷3)·Ⓒπ·x³' 'x' ∂`) rather than `4/3` (which parses as a fraction) also produces the wrong result.

A multiplication on the other hand produces the correct result: `''(4·3)·Ⓒπ·x³' 'x' ∂` yields `'36·π·x²'`.

commit 9b5497e2 (2025-05-22): derivative: Evaluate what matches constants in patterns

When a pattern contains a constant, e.g. `zero*X`, we need to evaluate
what matches `zero` to check if the *value* matches.

Fixes: #1484
commit aa2fd85b (2025-05-23): derivative: Evaluate what matches constants in patterns

When a pattern contains a constant, e.g. `zero*X`, we need to evaluate
what matches `zero` to check if the *value* matches.

Fixes: #1484