c3d/db48x #1485: Error of units management in a program that crashes the simulator

Wiljea, 2025-05-17:
This program (run with DB48x 0.9.6-12-5d503) calculates the value of a function and its uncertainty, the latter being obtained using the quadratic superposition formulae :  
     σf = √(|∂f/∂x|²·σx²). 
The program succeeds when there is no units for x, but fails otherwise:
```rpl
« 0 0 0 → f xv σx funits σxunits fv « 'f|x=xv' Eval →Num 'fv' STO fv IF fv Typename "unit" == THEN 1 fv →Unit ELSE 1 END 'funits' STO f 'x' ∂ 'x=xv' | Eval →Num abs IF σx Typename "unit" == THEN 1 σx →Unit ELSE 1 END 'σxunits' STO IF σxunits 1_° == THEN σx →Radians Uval 'σx' STO END σx × CASE funits 1_° == THEN 1_r * →Degrees END funits 1_r == THEN 1_r * END END » » 'Ssf' STO
'exp(-x/2)·sin(0.5·x)' 2 0.05 Ssf
@ Expecting [ 0.30955 98756 53 0.00276 98441 33 ]
'exp(-x/2_s)·sin((0.5_r/s)·x)' 2_s 0.05_s
@ Failing [ 0.30955 98756 53 0.00276 98441 33 ] => Simulator crash
```
The simulator crashes directly but looking at the (Fedora-Remix) console, it seems that the error starts with "Inconsistent units". Otherwise, the program is able to manage correctly units in simpler case as in the following:
```rpl
'4·Ⓒπ·x²' 2_m 0.05_m Ssf
@Expecting [ 50.26548 24574 m↑2 2.51327 41228 7 m↑2 ]
```

evgaster, 2025-05-17:
Ok, it shouldn't crash.

If x has units, what would the units of exp(x) be?
I think the argument to a function like exp() should not have units.

Wiljea, 2025-05-17:
Indeed, the whole argument, 'x' in 'exp(x)', 'ln(x)', ... shall never have units... just think at the incompatible mess that would result in the Taylor series expansion ! That's why here, I divided by 2_s in exp(x/2_s) if I have x_s.

Looking just at the derivative without evaluating it, leads to an error when "a" & "b" are numbers of type "unit":
'exp(-x/b)·sin(a·x)'  'x' ∂ => 'exp(-(x÷b))·a·cos(a·x)+(-(b÷b²))·exp(-(x÷b))·sin(a·x)' 
'exp(-x/2_s)·sin((0.5_r/s)·x)'  'x' ∂ =>"x² error: Inconsistent units"
=> '0.5 r/s+(-((0.5 r/s·x-A)÷2 s²))·exp(-(x÷2 s))·sin(0.5 r/s·x)'

c3d, 2025-05-20:
At least on the latest simulator, I cannot crash until I add `Ssf`.

So the code for the first one is:

``` rpl
« 0 0 0 → f xv σx funits σxunits fv « 'f|x=xv' Eval →Num 'fv' STO fv IF fv Typename "unit" == THEN 1 fv →Unit ELSE 1 END 'funits' STO f 'x' ∂ 'x=xv' | Eval →Num abs IF σx Typename "unit" == THEN 1 σx →Unit ELSE 1 END 'σxunits' STO IF σxunits 1_° == THEN σx →Radians Uval 'σx' STO END σx × CASE funits 1_° == THEN 1_r * →Degrees END funits 1_r == THEN 1_r * END END » » 'Ssf' STO
'exp(-x/2)·sin(0.5·x)' 2 0.05 Ssf
@ Expecting [ 0.30955 98756 53 0.00276 98441 33 ]
```

This works with `RAD` as the unit, and you need to also collect the two results as a vector to match `@ Expecting`:

``` rpl
« 0 0 0 → f xv σx funits σxunits fv « 'f|x=xv' Eval →Num 'fv' STO fv IF fv Typename "unit" == THEN 1 fv →Unit ELSE 1 END 'funits' STO f 'x' ∂ 'x=xv' | Eval →Num abs IF σx Typename "unit" == THEN 1 σx →Unit ELSE 1 END 'σxunits' STO IF σxunits 1_° == THEN σx →Radians Uval 'σx' STO END σx × CASE funits 1_° == THEN 1_r * →Degrees END funits 1_r == THEN 1_r * END END » » 'Ssf' STO
RAD
'exp(-x/2)·sin(0.5·x)' 2 0.05 Ssf
→V2
@ Expecting [ 0.30955 98756 53 0.00276 98441 33 ]

```

Wiljea, 2025-05-20:
I agree that the program ssf succeeds when there is no units for x, but fails when there are units.

c3d, 2025-05-20:
To get a crash you need:

``` rpl
«
    0 0 0
    → f xv σx funits
    σxunits fv
    «
        'f|x=xv' Evaluate →Num
        'fv' Store fv
        if
            fv TypeName "unit"
            ==
        then
            1 fv →Unit
        else
            1
        end 'funits' Store
        f 'x' ∂ 'x=xv' |
        Evaluate →Num abs
        if
            σx TypeName "unit"
            ==
        then
            1 σx →Unit
        else
            1
        end 'σxunits' Store
        if
            σxunits 1_° ==
        then
            σx →Radians UnitValue
            'σx' Store
        end
        σx ×
        case
            funits 1_° ==
            then
                1_r × →Degrees
            end
            funits 1_r ==
            then
                1_r ×
            end
        end
    »
»
'Ssf' Store

RAD
'exp(-(x÷(2_s)))·sin((0.5_r/s)·x)'
2_s 0.05_s Ssf →V2
@ Failing [ 0.309559875653 0.002769844133 ] => Simulator crash
```

Wiljea, 2025-05-22:
I don't understand the "DONE" comment, because this is still crashing the simulator in my case.

commit c5b68d58 (2025-05-22): cleaner: Clear command when in temporaries

When we put a command that is in the temporaries area, we need the
cleaner class to clear the command.

In the example given in #1485, we parse a temporary from a unit object
that is then in the purged area, but not above the end result.

Fixes: #1485
closed 2025-05-26 by c3d