c3d/db48x #1493: requesting optional color scheme …

beneliath, 2025-06-07:
- [ ] a la HP48SX (dark brown background with white keys, orange shift, and blue double shift)

beneliath, 2025-06-07:
for clarification, i meant for the emulator

prismv, 2025-06-07:
Okay - this was not very clear, because this would be a variant of the _simulator_ with a key layout not fitting to DM42x or DM32.

Indeed it's interesting, if/when "The project in this repository is presently targeting the SwissMicro DM42 and DM32 calculators." will change.

doug gilmore schrieb am Samstag 07 Juni 2025 16:37:33 (+02:00):

 [Unbeschriftetes Bild]
beneliath
 left a comment 
(c3d/db48x#1493)
 <https://github.com/c3d/db48x/issues/1493#issuecomment-2952586683> 

for clarification, i meant for the emulator

beneliath, 2025-06-07:
specifically (and where otherwise generalizable) the iphone app as an optional color scheme

c3d, 2025-06-07:
@beneliath Can you modify all the [existing picture](https://github.com/c3d/db48x/tree/stable/DB48X-Keys) to match your desired color scheme and submit that as a PR?

beneliath, 2025-06-08:
👍 of course. happy to do that, but difficult to mod rastered png; can you post resource vector file(s) used to render *.png?

### proposed color scheme:
|object(s)|hex color|described color|
|--|--|--|
|general background|`#403938`|charcoal brown, smoky umber|
|keypads & field box label backgrounds ("MAIN", "ALGB", _etc._)|`#221815`|dark roast, charred umber|
|white text on keys|`#E5E0C8`|pale khaki, sandstone|
|orange (single shift)|`#B8763A`|burnt sienna brown, golden ochre|
|blue (double shift)|*`#8BB1C3`|dusty sky blue, muted steel blue|

_*blue text over 48SX keys is a little more electric/bright than the color of the right-shift button itself and may need to be brightened_