c3d/db48x #1512: Constant ∞ multiply by a quoted expression: Bad argument type

LdBeth, 2025-07-30:
As of 0.9.11, a statement like `∞ '12+43' *` gives type error. Oddly enough, this does not happen with adding/subtracting.

Also a subtle thing with expression is `'Ⓒ∞+2' →Num` evaluates to a `∞` symbol that is distinct from `Ⓒ∞`, which I'm uncertain if is intended behavior.

LdBeth, 2025-07-30:
Can confirm this is related to the AutoSimplify flag.

LdBeth, 2025-07-30:
I'm certain this is caused by
https://github.com/c3d/db48x/blob/adbd0fb439f27908d01aecaaa39a66626d62230c/src/arithmetic.cc#L596

where the expression data type misses an implementation for `is_negative()` and raises a type error.

LdBeth, 2025-07-30:
This also implies there are some additional special cases not covered:

* `∞ "Text string" -` gives `∞` instead of type error when AutoSimplify on.
* A proper check that does not consume arguments should be implemented for `∞ "Text" *`.
* `optimize<struct pow>` code for cases involves expression

LdBeth, 2025-07-30:
So I compared with the actual behavior on HP50g while trying to implement some fixes, and noticed some other areas requirement clarification or still needs improvements:

Seems HP50g would always try to evaluate computation involves constant `∞`, regardless of "Simp non-rational" setting in CAS, I think it actually always try to get the sign of an expression, such as `'x+4' ∞ *` would produce `?`. The current db48x implementation does not check AutoSimplify in power operation but check that flag in other operations when dealing with `∞` which is an inconsistency.

The current simplifier does not yet properly simplify or produce the `?` (undefined result) for things like `∞-∞` `(-1)^∞`, until then it is hard to figure out what should be the desired behavior in `optimize<Op>` methods.

c3d, 2025-10-21:
Thanks @LdBeth for the analysis. How to properly simplify infinities is a complex topic, I will defer that (and #1513) to a later release (after 0.9.12).

c3d, 2026-02-08:
Fixed on `dev`, closing.

closed 2026-02-08 by c3d