c3d/db48x #1515: Error in uncertain range function "→σRange"

Wiljea, 2025-08-01:
If I do:
```rpl
3 4 →σRange
@ Expecting 3±σ4
```
which is ok. But, if I do:
```rpl
4 3 →σRange
@ Failing 4±σ3 => 3±σ4
```
which is a mistake. Why the function "→σRange" is reordering its input? There is no reason.

evgaster, 2025-08-03:
[The →σRange documentation](https://github.com/c3d/db48x/blob/stable/doc/commands/ranges.md#%CF%83range) seems incomplete en contains "( a±b%)". That needs improving.

The other "→*Range" commands have a remark "Input values are sorted ..". Perhaps that is done for →σRange as well. 

If the 2 arguments for the command are in order, it works:
```
-100 3 →σRange
-100±σ3
```
A peek at the C++ code (which I certainly do not fully understand) indicates that for  a σRange the arguments are not sorted. But your (and mine) test show otherwise.

Wiljea, 2025-08-03:
I think you've nailed it !

evgaster, 2025-08-03:
In my local repository clone I changed the unconditional sort to:
```   
if(! (ty == uncertain::ID_uncertain))
    range::sort(lo, hi);
```
That fixes the issue.

1. Haven't checked for any collateral damage;
2. Tried without "uncertain::" first. That caused a compile time error about. Putting the "uncertain::" in is a blunt ax move due to insufficient C++ and DB48X code knowledge. @c3d may have a better fix.
3. Don't know yet (insufficient git knowledge) how to get this into development code base. @c3d any hints on how you want me to do that?

Anyway, feels great to make this small contribution to this fantastic project.

LdBeth, 2025-08-03:
@evgaster According to existed code the namespace for ID's is `object::ID_*`, otherwise I think the fix is fit.

Wiljea, 2025-08-04:
So, we have to wait for @3d's intervention to make your fix in production in the dev branch. Thanks to both of you.

evgaster, 2025-08-07:
Got a better understanding of git and gtihub. Managed to get as far as creating a [pull request](https://github.com/c3d/db48x/pull/1516) for my fix for this issue.

That should make it easier on @c3d.

closed 2025-10-22 by c3d