c3d/db48x #1517: unclear state of UDEFINE and UPURGE commands

evgaster, 2025-08-09:
Working with the simulator; version 0.9.11

doc/commands/units.md mentions UDEFINE and UPURGE. But how to use them isn't described.
doc/6-ImplementationStatus.md doesn't mention them.
Can't find them in the CAT(alogue) in the simulator.
Looking for them in te source code gives me no hits.
Can't find them in "HP 50g Advanced users reference manual", "HP 50g users manual" or "HP 50g users guide".

You can "create" new units like 3_newunit just by "typing" them. But I don't see (yet) how to build one from SI units. I guess that's the intend of UDEFINE.

Reviewing [the units documentaton](https://github.com/c3d/db48x/blob/stable/help/db48x.md#units) maybe provides a workaround for not having UDEFINE.

If the intend is to implement UDEFINE and UPURGE, I could propose documentation, tests ...

LdBeth, 2025-08-10:
It is because the documentation were automatically generated from NewRPL, however, they are not implemented yet.

https://newrpl.wiki.hpgcc3.org/doku.php?id=manual:chapter3:units

LdBeth, 2025-08-10:
It was a one time thing. But if you want to submit changes, please note that help files in `help/` are built using files from `doc/`

evgaster, 2025-08-13:
Provided a workaround in the documentation.
Created a pull request for it.

evgaster, 2025-08-17:
Just noticed that my change to the example for →Unit makes a test "Enter example from help file into command line" fail.

See if can fix that.

evgaster, 2025-08-20:
Could almost fix it. But I don't know how to get the right screenshot.
Decided to backtrack on the cosmetic change I made.

There is a corresponding commit in the pull request.