c3d/db48x #1518: feature request: default correlation ρ=1 for identical variable names X in expression evaluations (X is ID_uncertain)

prismv, 2025-08-09:
Libraries dealing with measurement uncertainty calculations usually assume full correlated variable for identical names, so that (e.g.):

- X - X = 0.0 ± 0.0
- X/X = 1.0 ± 0.0
- X * √X is equivalent to X^(3/2) also for ID_uncertain type
- ...

See [db48x Thread post \#804](https://www.hpmuseum.org/forum/thread-20157-post-207968.html#pid207968) up to [\#810](https://www.hpmuseum.org/forum/thread-20157-post-208021.html#pid208021).

Wiljea, 2025-08-09:
It is important to distinguish for which type of Range you do these calculations and comparisons, considering the fact that DB48x treats mainly two types of Ranges: 

1. the **interval** type: **a…b** and its two other equivalent forms: the **drange** type: **a±b** & the **prange** type: **a±b%**. Remember that you can always cycle ("o" on the keyboard) between them. These 3 types are all treated in the same way for internal computations and they all obey the basic principles of interval arithmetic with its famous dependency problem that I will describe below (see https://en.wikipedia.org/wiki/Interval_arithmetic).

2. the **uncertain** type: **a±σb** which is the only one treated by quadratic superposition of uncertainties, generalised here to include the covariance σab=ρab·σa·σb which depends on the correlation coefficient ρab between the normal random variables a & b (see for instance: https://en.wikipedia.org/wiki/Propagation_of_uncertainty#:~:text=.-,Non%2Dlinear%20combinations,-%5Bedit%5D). The important point is that for a function f(a, b) of 2 random variables a & b, the resulting uncertainty can be approximated (to 1st order) by:
     σf ≈ √( |∂f/∂a|²·σa² + |∂f/∂b|²·σb² + 2·(∂f/∂a)·(∂f/∂b)·ρab·σa·σb )  
     where  ρab·σa·σb=COVAR(a, b) 
and this can be extended for more than 2 variables using a symmetric covariance matrix (see: [https://en.wikipedia.org/wiki/Experimental_uncertainty_analysis#:~:text=on%20data%20analysis.-,Approximation%20for%20the%20variance%20of%20z,-%5Bedit%5D](https://www.google.com/url?q=https://en.wikipedia.org/wiki/Experimental_uncertainty_analysis%23:~:text%3Don%2520data%2520analysis.-,Approximation%2520for%2520the%2520variance%2520of%2520z,-%255Bedit%255D&sa=D&source=docs&ust=1754744955441298&usg=AOvVaw1MNgB7xIXo0vKZLNV8UZ2O))

For instance, as far as I can see for the **uncertain** type of Range (the only one concerned with the concept of correlation), if you choose ρ=+1, you will obtain precisely what you are expecting. So there should be no issue if you compare with external packages treating the variables in the same framework.

Otherwise, if you are comparing with an external package performing standard interval arithmetic (or using case 1, here in DB48x), the dependency problem will give entirely different results since it considers each occurrence of random variables in an expression as being totally independent. This problem is well known and can lead to an unwanted expansion of the resulting interval. Various techniques are possible to minimise overestimation (dependency tracking, mincing, splitting, single occurrence of variables, etc.).

prismv, 2025-08-10:
Yes, the problem can be reformulated that "Obviously, the same uncertainty variable occurring multiple times in the same algebraic expression is considered identical;" is currently not the case, if this variable is of type ID_uncertain.

If I'm not mistaken the error propagation result is the same for identity or full correlation, i.e. autocorrelation.

Wiljea, 2025-08-10:
Yes, ok with me.