c3d/db48x #1532: GammaLn crash

emb974, 2025-08-31:
enter 2 Menu Prob Ln(Gamma) results in crash
r0=00000000 r1=00000000
r2=40017867 r3=00000000
r12=000000b4 lr=00000003
pc=0815f240 psr=81000000

[image]

Wiljea, 2025-08-31:
On what version of the firmware do you experience a crash? For me, in version of the simulator DB48X 0.9.11-3-f362 (constructed from dev), I don't see any crash:
```rpl
8.0 lgamma @Expecting 8.52516 13610 7
8.0 gamma  @Expecting 5 040.
5 040. ln @Expecting 8.52516 13610 7
```
Everything seems ok for me. The function lgamma(x) being equivalent to 🟨 × 🟨 F2.

Wiljea, 2025-08-31:
> Dear Wiljea, thank you very much for responding - and so quickly!
> Please excuse me for being not very exact, I am not proficient in handling
> errors but I follow this fantastic project and wanted to help with my
> observation. Maybe I didn't configure something correctly or there is
> perhaps a nomenclature clash after my last upgrade to 0.9.11 two weeks ago.
> Before the summer it worked ok in some little RPL routines for statistical
> functions in my library.
>
> If this cannot be reproduced you can close it. In any case I replaced it
> with a short RPL.
>
> Thanks again and kind regards
>
> 20250831_182214.jpg (view on web)

@emb974
You're welcome. I will keep the ISSUE open for a couple of days, in case
you're not alone with this problem.

emb974, 2025-09-01:
Dear Wiljea,
it is true that using the Demo mode it worked ok, so I checked the altered options that caused the crash: It turned out to be Hardware Floating Point. (I have some routines that work with this...)
Kind regards

emb974, 2025-09-01:
(just for the record: I had precision setting of 12 or lower to enable HWFP. If it is e.g. 24 HWFP will not be active and LogGamma works fine - results are displayed with D (double precision) in HWFP )

c3d, 2025-10-05:
I need to look into this. This rings a bell, so I may have fixed it somewhere already.

emb974, 2025-10-15:
For a humble layman it is just amazing to see how the bits are shuffled around in my little calculator... Please let me express deep thanks for this great project, and my admiration for your capability of this deep digging into the functioning on the processor level, it is just amazing. 
Thanks again!