c3d/db48x #1541: Test #7: istack: Interactive stack operations, 7:024.001: Interactive stack info about 5 fails

sneakywumpus, 2025-09-18:
While running the test suite, the following test failed on my system (Arch Linux, gcc 15.2.1 20250813):
```
../src/tests.cc:1416:  Test #7: istack: Interactive stack operations
../src/tests.cc:1416:   7:024.001: Interactive stack info about 5
../src/tests.cc:1416:    Expected screen to match [istack-18]
```

I ran the test case manually, and the hash value in the "Object info" screen had random characters following the 'h' base character.

I looked at the code, and "Object info" uses `draw_message`, which expects zero-terminated strings, but the rendered hash value isn't. The following change fixes this for me:

```
diff --git a/src/user_interface.cc b/src/user_interface.cc
index ecb41641..0923be46 100644
--- a/src/user_interface.cc
+++ b/src/user_interface.cc
@@ -4633,7 +4633,8 @@ bool user_interface::handle_editing(int key)
                         char valbuf[40];
                         snprintf(sizebuf, sizeof(sizebuf), "%lu bytes %s",
                                  (unsigned long) size, obj->fancy());
-                        bin->render(valbuf, sizeof(valbuf));
+                        size_t len = bin->render(valbuf, sizeof(valbuf) - 1);
+                        valbuf[len] = '\0';
                         draw_message("Object info", sizebuf, valbuf);
                         wait_for_key_press();
                     }
```

sneakywumpus, 2025-09-26:
I cleaned up the pull request as #1545 .

c3d, 2025-10-05:
@sneakywumpus Good catch, thanks a lot.

I picked it up today on my dev branch. Sorry for the long delay (I'm currently moving, so everything is a bit on hold).

commit cf1ff58f (2025-10-06): istack: Fix display of checksum in the `Info` box

`draw_message()` expects zero-terminated strings, the rendered
hash value in `valbuf` was not and could cause random characters
to appear after it, which led to failing the
"istack: Interactive stack info about 5" test.

Fixes: #1541